trezor/trezor-firmware #2721: Support for BIP127 proof-of-reserves

ulrichard, 2023-01-06:
**Design:** 
It would be really cool, if Trezor supported BIP127 proof-of-reserves. 
https://github.com/bitcoin/bips/blob/master/bip-0127.mediawiki

When I tried to sign a proof PSBT, I got the following error: 
`Segwit input without amount`
which might come from 
https://github.com/trezor/trezor-firmware/blob/master/legacy/firmware/signing.c#L2672

One of the special features of proof PSBTs is that the first input is a hash of a message with an empty amount.

I tried to sign it with a modified version of bdk-cli
https://github.com/bitcoindevkit/bdk-cli/pull/135
roughly following the steps described in
https://ulrichard.ch/blog/?p=2587 and https://ulrichard.ch/blog/?p=2566

The PSBT that I tried to sign is:

`cHNidP8BANABAAAABMErBiXTxwDcvvvq771RVgP1SYJh3scx0WBImDi7rvmsAAAAAAD/////ociiMb0zyfvtnyfSD/tL3j9+uhOjN/neZ3a0csAvgnUAAAAAAP/////OcQ+x1xeUR6g6qq5m5Lkh5VZFqRq/AEJMW19BnljCzAEAAAAA/////0dk+dyXtMMLgguLAeMt9N3+cdbF4uu0m7eRh8jSWEi/AQAAAAD/////ARY4AwAAAAAAGXapFJ9/0JbTftLA4/fwz8kkvu9P/OtoiKwAAAAAAAEBCgAAAAAAAAAAAVEBBwAAAQErZooBAAAAAAAiACDK4FhviqZnk06j1rP6xzX0ZIZVTNrduit4ibBbTCp28wEFR1IhAiP04QyJ0PS/reCKzBpL/Nq0bkuou3+OrBq4guz417nsIQK9ACGVMIAKE6Gvd2ixOvdLexWYP+iCMsO25eCLhWKE8VKuIgYCI/ThDInQ9L+t4IrMGkv82rRuS6i7f46sGriC7PjXuewMw7inHwAAAAACAAAAIgYCvQAhlTCAChOhr3dosTr3S3sVmD/ogjLDtuXgi4VihPEMRyCUewAAAAACAAAAAAEBKxAnAAAAAAAAIgAg+sR5av8FqkuQsGDYuwTOWM4/cj3QY3O/zHK+XlszvEQBBUdSIQJIO6aTsd07uVM4YwE78HlvGyPyqn8vZ/dAnPM0v5a2wyED751WGu3u7C5+GMlw2TX42HXDiKmNuMKdbN9wfAlKRIJSriIGAkg7ppOx3Tu5UzhjATvweW8bI/Kqfy9n90Cc8zS/lrbDDEcglHsAAAAAAQAAACIGA++dVhrt7uwufhjJcNk1+Nh1w4ipjbjCnWzfcHwJSkSCDMO4px8AAAAAAQAAAAABASughgEAAAAAACIAID9ZHeaqYFKSdaWxTi0XYWKsUjj0CcEn9EXdjww5W+PpAQVHUiECPNaDgf4tLQMy++nM84EbzaGc7cXYQfWKPQN+dSNfDN4hArgnsTM0wU3lmkxL2dkuqRIieVlZj5+/1G01p4939k3sUq4iBgI81oOB/i0tAzL76czzgRvNoZztxdhB9Yo9A351I18M3gzDuKcfAAAAAAMAAAAiBgK4J7EzNMFN5ZpMS9nZLqkSInlZWY+fv9RtNaePd/ZN7AxHIJR7AAAAAAMAAAAAAA==`

matejcik, 2023-01-06:
> When I tried to sign a proof PSBT, I got the following error:
> `Segwit input without amount`

This just indicates a bug in the signer application. I don't think we check that the input is non-zero, but the amount, even if zero, _must_ be a part of the incoming protobuf message.

A bigger problem will be that using Segwit, you will be required to provide a preimage for the prevtx hash that is (a) a valid Bitcoin transaction and (b) the selected outpoint matches the amount. There is no reasonable way around that, I'm afraid, certainly not one that would allow Trezor to sign the input.

This is no longer a requirement with Taproot however, so you might be able to do this if all your inputs are Taproot.

---

of course unless there is native support on Trezor, the user experience will be bad: to the user, it will, at best, look like a zero-BTC zero-fee transaction with no destination (if the single output is marked as change), or a X BTC transfer to some address (if the output is non-change)

prusnak, 2023-01-07:
BIP370 (PSBTv2) addressed lots of shortcomings of BIP174 (PSBT) with respect to hardware wallets.

I think the same needs to happen with proof of reserves BIP. A new BIP built on top of BIP370 that addresses shortcomings of BIP127 with respect to hardware wallets need to be created. There are extra fields in PSBTv2 so you can indicate to the hardware signer you are doing proof of reserves and not regular signing. 

Until this is done, I don't think it makes a lot of sense for us to pursue this, because we will never achieve great user experience and this has always been our goal.

closed 2023-01-09 by hynek-jina