trezor/trezor-firmware #2725: Prepare FW for bluetooth integration

Hannsek, 2023-01-09:


TychoVrahe, 2023-03-31:
This is the main branch i use https://github.com/trezor/trezor-firmware/tree/tychovrahe/bluetooth/master

Some of the issues above i already attempted to solve, i.e.:
- when USB workflow is in progress, and message is received via another interface, device response with busy: https://github.com/trezor/trezor-firmware/commit/1288f80c3b364d3e86a133597bd53ccc0bf3c824
- conditional inclusion of BLE: https://github.com/trezor/trezor-firmware/commit/6b911305b26b677a65e187c0d465d5409160345a

Building on https://github.com/trezor/trezor-firmware/pull/2912 which should be merged first.

Some other issues that will need addressing before we can merge this:
- nRF SDK is currently in embed/sdk/nrf52. I think we should make it a submodule. Bonus points: lets make a STM32 SDK/HAL submodule too and use that separately from the one in micropython. We will have same structure for all MCUs and be less dependent on micropython
- this also contains support for STM32 discovery, but the support is not compatible with the rest of codebase, we will need to either revert this or finalize it. I solved some issues with this while working on BLE, but some still remain. (personally i think supporting the discovery is useful for development and would be worth the effort)
- trezorctl bluetooth support needs imporvements: currently it always does scanning which causes annoying delays when issuing commands. and it also finds devices that are not reachable which causes malfunctions.
- some trezorctl dependencies used for bluetooth are not actually listed as dependencies

TychoVrahe, 2023-04-04:
- current impl in trezorctl is linux only

closed 2026-10-29 by obrusvit