trezor/trezor-firmware #2728: Use lower case in the name `coinjoin`

hynek-jina, 2023-01-10:
Replace word `CoinJoin` with `coinjoin` wherever it appears.

Same applies if we use coinjoin as a verb or noun.

bosomt, 2023-01-23:
**QA OK**

 - Device: model T 2.5.4 Universal (revision ab65b18b07170c8a9e4f1506620af988d678453b)

bosomt, 2023-01-23:
**QA NOK**

https://github.com/trezor/trezor-firmware/blob/ab65b18b07170c8a9e4f1506620af988d678453b/core/src/trezor/ui/layouts/tt_v2/homescreen.py

[image]

**Info:**
 - Suite version: desktop 23.2.0 (47ff6c1b0068967d941c67675a9b6cc012464fb7)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/23.2.0 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision ab65b18b07170c8a9e4f1506620af988d678453b)
 - Transport: bridge 2.0.32

hynek-jina, 2023-01-23:
@bosomt This doesn't look like a mistake. It's a sentence... so it starts with capital `C`, right? Or what is the problem?

bosomt, 2023-01-23:
@hynek-jina not for me :)
+ dialogue headers are all caps so it says ` "AUTHORIZE COINJOIN",`
if its OK feel free to close the issue

https://github.com/trezor/trezor-firmware/blob/1359b21508c7efd1038beb921dfc3516a4a07967/core/embed/rust/src/ui/model_tt/layout.rs#L938

hynek-jina, 2023-01-23:
Yes, I think it is intended.

closed 2023-01-23 by hynek-jina