trezor/trezor-firmware #2734: UI2 - not cleaning passphrase row properly (for a dollar sign)

grdddj, 2023-01-11:
By a coincidence, I have found out that the dollar sign `$` seems to have a bigger height than any other symbols available in `passphrase`.

It causes one upper pixel of dollar sign to be outside of the area belonging to the passphrase row. When deleting the passphrase, these pixels still stay there:

[image]

[image]

It is a very unique situation, maybe not even worth fixing, but we might want to know about this

matejcik, 2023-01-11:
i believe this is an instance of #2657

mmilata, 2023-03-02:
note to self: @grdddj has click test for this in the R branch, perhaps cherry-pick it when this is fixed

PR #2876 (2023-03-23): TT UI: style update
Fixes #2849, fixes #2734 and possibly #2657.

[image] [image] [image] [image] [image] [image] [image] [image] ![fc7c6eb03347115fec4de5f255f16eb580d2b12f60aab3ec01506d0cd2289bb6](https://user-images.
closed 2023-03-23 by mmilata