trezor/trezor-firmware #2737: T2B1: homescreen flickering when Suite loading balances

grdddj, 2023-01-12:
When Suite is asking `Trezor` for data about its addresses, the homescreen experiences flickering, even on emulator

grdddj, 2023-01-26:
No longer an issue (on emulator nor on hardware), most probably solved by a recent rebase on `master` which included some new code for handling the homescreen

closed 2023-01-26 by grdddj

Hannsek, 2023-01-26:
I can still see it on the emulator.
[image]

grdddj, 2023-01-26:
> I can still see it on the emulator. [image]

It is possible that you are using an old emulator - based on the job ID - `358...`. Unfortunately, newest pipelines do not build `ARM` emulator versions, there are some problems on the runners

Hannsek, 2023-01-26:
This should be the newest one according to emulator.

grdddj, 2023-01-26:
> This should be the newest one according to emulator.
> 
> Will the problems be solved?

Not sure, that is the question for adminkos - @vdovhanych