trezor/trezor-firmware #2755: Non-breaking space

TychoVrahe, 2023-01-19:
Introduce non-breaking space character, which will be used to prevent breaking lines where non appropriate, like after the `a` article.

andrewkozlik, 2023-01-26:
I was thinking about this just recently, when I didn't like breaking "[...] account #1" after "account".

I was looking briefly at the algorithm at the time and one thing I wanted to delve deeper into is how it breaks words. It introduces a dash at the end of the line, when it wants to break, but what if the word already has a dash at that position? For a word like "non-private" you could get:
```
[...] non--
private [...]
```
or
```
[...] non-
-private [...]
```
(This is untested, just my hypothesis.)

Finally, I was thinking about introducing something like `\-` in LaTeX, but the idea is slightly different. It would encourage the algorithm to break at that point if the whole word does not fit on the line. So for example, let's say you have a sentence with words "{W1} {W2} {W3} ..." and W2 is `every\-where`, then:
- if "{W1} everywhere" fits on a line, then ignore the `\-` and continue as usual.
- if "{W1} everywhere" doesn't fit on a line, but "{W1} every-" does fit, then break it as ["{W1} every-", "where {W3} ..."].
- if "{W1} every-" doesn't fit on a line, but "{W1}" does, then break it as ["{W1}", "everywhere {W3} ..."].

We'd probably want to reuse some of the standard escape-sequences for the nbsp and the `\-`. The options seem to be `\t`, `\b`, `\f`, `\v`, `\a`.

mmilata, 2024-06-06:
Looks like we started receiving translated strings containing non-breakable spaces in the right places, yet can't handle them.

* we could add the char to `core/tools/codegen/foreign_chars.py` but we'd need to introduce similar mechanism for english which probably doesn't make sense, or
* we could add special handling for 0xA0 codepoint in the text renderer.

After it works the commit 1b31b4033069fc0d4d789e5e3a004b5c8cfc9199 needs to be reverted.

matejcik, 2024-06-06:
> we could add special handling for 0xA0 codepoint in the text renderer.

I'd just hack `get_glyph` to return the 0x20 (ascii space) glyph for 0xA0 also. That way the space is counted for width but doesn't count as a breaking point (those are hard-coded to ascii space and newlines i think?)

PR #4010 (2024-07-30): feat(core): deal with non-breaking spaces
Fixes #2755
closed 2024-07-30 by ibz