trezor/trezor-firmware #2765: Upload of firmware to devkit is failing on macARM

bosomt, 2023-01-24:
I'm not able to upload bigger firmware to devkit since we received it.
Works as expected on Windows 10 or on KUbuntu.
It always freezes during transfer 

```
trezorctl version
0.13.6

 ~/Downloads/trezorfwasuite/devkitfw  trezorctl firmware-update -f trezor_1.6.1_bld_1.4.0.bin -s      
Uploading  [------------------------------------]    0%
Update failed: trezord: post/46 failed with code 400: hidapi: hid_error is not implemented yet

 ~/Downloads/trezorfwasuite/devkitfw  trezorctl firmware-update -f trezor_1.6.1_bld_1.4.0.bin -s           
Uploading  [------------------------------------]    0%
Update failed: trezord: post/49 failed with code 400: hidapi: hid_error is not implemented yet

 ~/Downloads/trezorfwasuite/devkitfw  trezorctl firmware-update -f trezor_1.5.1.bin -s                    
Uploading  [------------------------------------]    0%
Update failed: trezord: post/52 failed with code 400: hidapi: hid_error is not implemented yet
```

 - Suite version: desktop 23.2.0 (4c8e4dccd95e1ab4c85bc1ad507c114e1b02fd48)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/23.2.0 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision ab65b18b07170c8a9e4f1506620af988d678453b)
 - Transport: bridge 2.0.32

prusnak, 2023-01-24:
I don't think this is about the firmware size, but rather about HID in the bootloader. Which version is the bootloader?

bosomt, 2023-01-25:
@prusnak looks like you are right about that, checked it with different bootloaders and it happens only with old `1.4.0` and not with `1.12.x` bootloader. 

I will update this issue during next round of bootloader testing

bosomt, 2023-02-01:
@prusnak you are right only old images with old bootloader like `trezor_1.6.3_bld_1.5.1.bin` are failing

bosomt, 2023-04-04:
related only to old firmwares

closed 2023-04-04 by bosomt