trezor/trezor-firmware #2784: Trezor Model T does not work as an Apple Security Key

DavidVeksler, 2023-02-02:


bosomt, 2023-02-02:
@DavidVeksler 

can you please share more information how to navigate to add security section  ?
I was not able to find dd key in `Password and Security`  in iPhone or macOS v 13.1 even i have 2FA enabled

[image]

[image]

andrewkozlik, 2023-02-02:
The Trezor T implements FIDO2 and uses "self attestation" by default. Trezor 1 implements only the older U2F protocol, which supports only "basic attestation". So maybe the reason for this is that Apple accepts only basic attestation. We can override the default attestation for Apple devices on Trezor T, but I am unable to reproduce this issue.

DavidVeksler, 2023-02-02:
> can you please share more information how to navigate to add security section ?

This feature was just added in MacOS 13.2 and iOS 16.3

bosomt, 2023-02-02:
upgraded to 13.2
and i have to wait a little

[image]

Kl-11, 2023-02-16:
Hey @andrewkozlik - thanks for looking at this. Any updates? Facing the same issue here.

andrewkozlik, 2023-02-16:
It doesn't seem to be related to attestation. I prepared a firmware that uses basic attestation for someone from our company to test and they say it didn't work either.

My next best guess is that it could be related to the ClientPin feature, which is not supported on Trezor, because on Trezor you can enter the PIN on the screen, so there is no use for it. But maybe Apple requires it, simply because pretty much all authenticators on the market are single-button devices with no screen. Verifying this hypothesis would require trying to implement the feature, which I don't have the capacity for right now and the result is uncertain.

The most useful thing right now would be to see a transcript of the USB communication between MacBook/Trezor and between MacBook/Yubikey. On linux I can do that with wireshark.

Kl-11, 2023-02-17:
Thanks @andrewkozlik - does that still applies if it works with the other Trezor model? As I understood @DavidVeksler was saying (please correct me if I’m wrong).

andrewkozlik, 2023-02-17:
It still applies, because Trezor T uses the newer FIDO2 protocol, whereas the Trezor model 1 uses U2F.

DavidVeksler, 2023-02-17:
I tried to use Wireshark, but it cannot capture USB traffic in MacOS Ventura because USB is not listed as an interface.  I will share the log if someone can suggest a way to do that.

prusnak, 2023-02-17:
@DavidVeksler here's the wireshark guide for usb capture: https://wiki.wireshark.org/CaptureSetup/USB#macos

Apparently you also need to disable SIP on newer version of macOS (catalina and newer).

andrewkozlik, 2023-02-20:
1. Disable SIP: https://developer.apple.com/documentation/security/disabling_and_enabling_system_integrity_protection
2. Look at the list of interfaces `tcpdump -D` you should see something like `XHC0` or `XHC20` in the list.
3. Bring up one of the USB interfaces you saw in the list by executing `sudo ifconfig XHC0 up`.
4. Use wireshark to capture communication with a FIDO2 authenticator that works. (U2F communication such as Trezor One is not interesting in this case. FIDO2 Yubikey would be very interesting. Trezor T is less interesting.)

DavidVeksler, 2023-02-27:
Thanks @andrewkozlik .  Here is the log from Wireshark:

[image]

[add apple security key.pcapng.gz](https://github.com/trezor/trezor-firmware/files/10844874/add.apple.security.key.pcapng.gz)

DavidVeksler, 2023-02-28:
Thanks for looking into it.  Just want to emphasize that once I connect with my Model One, the Model T works fine to sign in. 

Would it help to capture Model One traffic for comparison?

andrewkozlik, 2023-02-28:
> Would it help to capture Model One traffic for comparison?

No. I am afraid not. The traffic is perfectly fine. Apple is not accepting the registration. It could be a bug on their side or an intentional restriction. It's just guesswork now.

However, I am not convinced that it's intentional given that the Model One works and given what they state on their site: "_If you choose a different security key, you should choose security keys that are FIDO® Certified_ [...]". They are not saying you have to use a certified authenticator.

I will need to think about whether we can somehow force fallback to U2F for Apple registration. It could also be that they don't like the size of the FIDO2 credentials, which are larger than the U2F credentials generated by Model One.

andrewkozlik, 2023-03-20:
I prepared a firmware which uses smaller credentials and a colleague of mine was able to use it to do a successful FIDO2 registration with Apple. Unfortunately, when he tried registering again a few days later, it gave him the error below and it did not even show the registration request on Trezor. That would seem to indicate that Apple might be refusing Trezor's AAGUID, which it didn't do before. @DavidVeksler can you try to register again with your Trezor T and tell me if the behavior has changed for you too?
[image]

andrewkozlik, 2023-03-20:
Update: @DavidVeksler  please ignore the previous comment. It looks like @vdovhanych is banned from registering any FIDO/U2F authenticators right now. I guess he was testing too much.

PR #2931 (2023-04-05): Enable Trezor to work as a FIDO2 authenticator for Apple.
Fixes #2784

Trezor T can't be used as an Apple security key, because the FIDO2 credentials are a bit larger than U2F credentials. Apple has some internal size limit on credentials and refuses to store them. As a result it's possible to use Trezor One as a security key with Apple, but not Trezor T. In this PR I introduce the `use_compact` option which reduces the amount of information stored in the credential to a minimum. After enabling this option for Apple, the FIDO2 registration works fine.

<s>Before merging @vdovhanych just needs  [... cut]
closed 2023-04-05 by andrewkozlik