trezor/trezor-firmware #2798: Find a way how to install/update new unreleased versions of trezorctl

Hannsek, 2023-02-07:
We need to find a way how to install version of `trezorctl` which is not yet publicly released. Testing reasons.

matejcik, 2023-02-07:
this is trivial. run it from the `poetry shell` from current checkout.

closed 2023-02-07 by matejcik

Hannsek, 2023-02-07:
This is too difficult for non-tech persons as e.g. designers.

matejcik, 2023-02-07:
and using trezorctl by itself isn't? :) where is the line? if `pip install trezor` is ok, would this work?
```
pip3 install "git+https://github.com/trezor/trezor-firmware.git#egg=0.13.5&subdirectory=python
```
(where `egg=0.13.5` is "what version you want to pretend this is")
if that works, why not this?
```
git clone https://github.com/trezor/trezor-firmware
cd trezor-firmware
poetry install
poetry shell
```
if that's too many commands, then `./install-trezorctl.sh` which has the above contents?

Hannsek, 2023-02-07:
> pip3 install "git+https://github.com/trezor/trezor-firmware.git#egg=trezor&subdirectory=python"
this is fine, thank you

is there a way how to uninstall it and install the old version?

matejcik, 2023-02-07:
`pip3 uninstall trezor` then `pip3 install trezor`