trezor/trezor-firmware #2799: `unix ui changes` job in CI fails when there are some changes

grdddj, 2023-02-07:
`python -m tests.ui_tests.reporting master-diff` seems to be failing in case there are some UI changes or new tests - both for `device` and `click` tests.

https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/3726810931 from https://github.com/trezor/trezor-firmware/pull/2763
https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/3723274057 from https://github.com/trezor/trezor-firmware/pull/2610

matejcik, 2023-02-07:
what's weird is that there is a visible stack trace for a failed download, but then it goes on to download more...?

matejcik, 2023-02-10:
fix, hopefully, in above linked commit.

it looks like some fixtures were not deployed properly, so we'll need to watch this space and see if the problem magically fixes itself -- otherwise we'll get bad UI diffs and need to look deeper into it

grdddj, 2023-02-14:
Looks like there is another "edge-case" that we do not account for - when there is a rename of the test-case and therefore the older `master` test-case cannot be found in the new test-cases.

Example: "deleting" `test_recovery` test-case because of creating `test_recovery_bip39` and `test_recovery_slip39` 

https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/3762985591

commit 0e9741f0 (2023-02-10): fix(tests): more robust master diff handling

fixes #2799
commit 26639702 (2023-02-10): fix(tests): more robust master diff handling

fixes #2799
commit 9add2b00 (2023-02-10): fix(tests): more robust master diff handling

fixes #2799
closed 2023-02-17 by grdddj