trezor/trezor-firmware #2805: Update Poetry

matejcik, 2023-02-08:
dependabot PRs completely break our CI because they're using a newer version of `poetry.lock`.

We can, and should, update our Poetry to avoid this problem in the future.

matejcik, 2023-02-27:
this is actually breaking HWI tests now

mmilata, 2023-03-02:
We do want the hashes there for reproducibility reasons, right?

Currently we have poetry-1.1.13, today nixpkgs have 1.3.2 with 1.4.0 on the way nixos/nixpkgs#218868.

matejcik, 2023-03-02:
> We do want the hashes there for reproducibility reasons, right?

absolutely, yes. i'm seeing some hashes discussion [here](https://github.com/python-poetry/poetry/pull/6393), do you know what that is about?

matejcik, 2023-03-02:
> reproducibility reasons

security reasons, strictly speaking. if some part of the supply chain is attacked, we don't want to pull in a different thing just because it is called by the same name.

mmilata, 2023-03-02:
> do you know what that is about?

No idea tbh. I'm asking because we used to have the hashes before 5196f249a214f35a8babc6dd7057cba5baa46d15, and running `poetry add freeetype-py` with poetry-1.1.3 adds them again, was wandering what's going on.

prusnak, 2023-03-07:
https://github.com/NixOS/nixpkgs/commit/2db50277cbfdd7600a37cc72b32e50438a08230b is the first nixpkgs commit that contains poetry-1.4.0

Trying it out ...

prusnak, 2023-03-17:
> Hopefully it propagates soon from `staging-next` to nixpkgs-unstable.

Poetry 1.4.0 in nixpkgs-unstable now -> created PR https://github.com/trezor/trezor-firmware/compare/nixpkgs-bump which bumps nixpkgs to latest unstable.

closed 2023-04-19 by mmilata