trezor/trezor-firmware #2812: T2B1 design - homescreen

Hannsek, 2023-02-10:
For now, let's use this one:

[image]

[image]

Figma link: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=4918%3A15122&t=T5ORQkrVVHFI8q8a-4

We need to see it on the device.

grdddj, 2023-03-25:
Designs of homescreen/lockscreen with logo done in [4fb4a74](https://github.com/trezor/trezor-firmware/pull/2610/commits/4fb4a74e500855e118760a330c393725e22ba416)

Current limitations/issues of homescreen and welcome screen:
- on hardware, the welcome screen is shown for a very short time before homescreen, most of the time there is black screen
- on hardware, on homescreen, sometimes `NO USB CONNECTION` is displayed for a short moment after calling e.g. `trezorctl set pin`
- homescreen currently does not support locking via button-hold

Hannsek, 2023-03-25:
Ad points 1 and 2. So it behaves differently than model T? Why?

grdddj, 2023-03-26:
> Ad points 1 and 2. So it behaves differently than model T? Why?

That's a mystery to me, there could be some code differences/bugs or hardware problems. Will troubleshoot it

Hannsek, 2023-05-09:
[image]

https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?type=design&node-id=8287-78653&t=F8FxwBHaV13SepO5-4

grdddj, 2023-05-10:
New designs done in [7dfc601](https://github.com/trezor/trezor-firmware/pull/2610/commits/7dfc60128359c2184cf542ce602403749ac18ab0)

[image]
[image]
[image]