trezor/trezor-firmware #2823: UI2: Welcome screens (uninitialized)

Hannsek, 2023-02-15:
[image]

Hannsek, 2023-02-20:
@matejcik The last screen is meant to be there as a homescreen between fw installation and the end of an onboarding process. 

Having there some link like `trezor.io/start` does not make much sense in this situation as you already have Suite installed.

matejcik, 2023-02-20:
again:

> The last screen is meant to be there as a homescreen between fw installation and the end of an onboarding process.

This does not match the idea that this screen communicates (from fw point of view at least) something like "please wait until your Trezor is fully started up".
Maybe we don't need that in the "not set up" case though?

> Having there some link like trezor.io/start does not make much sense in this situation as you already have Suite installed.

the user _can_ end up in a situation where they have FW installed, but didn't finish the setup, e.g.: fw installed on work PC, continuing at home; or continuing the next morning and forgetting everything

Hannsek, 2023-03-02:
https://github.com/trezor/trezor-firmware/issues/2825#issuecomment-1450179615

PR #2879 (2023-04-03): TT bootloader style update

Had to adjust backlight PWM setting a bit, to make transitions bootloader->firmware look good without flickering.

Cherrypicked fonts from https://github.com/trezor/trezor-firmware/pull/2876 for now.

The RSOD screens still use paragraphs, which i need for vertical centering that Label cannot do yet - but i believe that such functionality would be useful for Label as well a would simplify things here a bit.

Handles "uninitialized" boot flow as well. 

Closes #2866 
Closes #2823
closed 2023-04-03 by matejcik