trezor/trezor-firmware #2824: UI2: Coinjoin screens

Hannsek, 2023-02-15:
Will be this week…

Hannsek, 2023-02-16:
If the spinner is too complicated, you can use _loader non progress mode_

matejcik, 2023-02-16:
(i.e., that thing where you run a loader but don't tell it the total, so it just spins forever)

mmilata, 2023-02-23:
When are we supposed to show the homescreen notification (_COINJOIN RUNNING_)? AFAIK firmware is not currently aware of this?

Hannsek, 2023-02-23:
We should show this between coinjoin rounds. But if Trezor currently does not know that it is between rounds, let it be.

hynek-jina, 2023-02-23:
Trezor should know that it is allowed to sign some transactions without need of additional confirmation from user.

mmilata, 2023-02-23:
Ah, right, so it should be shown whenever `apps.common.authorization.get() == MessageType.AuthorizeCoinJoin` if I'm reading the code correctly.

hynek-jina, 2023-02-23:
> Ah, right, so it should be shown whenever apps.common.authorization.get() == MessageType.AuthorizeCoinJoin if I'm reading the code correctly.

@andrewkozlik can you please verify?

andrewkozlik, 2023-02-24:
Firstly there are two things to look at here:
1. Is coinjoin preauthorized? Then show "coinjoin running".
2. Is a coinjoin round in the critical phase? Then show "do not disconnect".

**Is coinjoin preauthorized?**
Doesn't `apps.common.authorization.get()` return the actual message? So you'd need to use `is_type_of()` or something like that. We could define `get_type()` to return the message type and check for `MessageType.AuthorizeCoinJoin`, which would avoid memory allocations to decode the message. But I see a bigger problem. The values returned depend on the active session. So if Suite switches the session to another seed/passphrase, then `apps.common.authorization.get()` will return `None`. What we need is something like `storage.cache.get_all(key: int) -> list[bytes]` which will return the key value across all sessions.

**Is a coinjoin round in the critical phase?**
`base.busy_expiry_ms() > 0`

andrewkozlik, 2023-02-24:
Like this https://github.com/trezor/trezor-firmware/commit/e8ac548eb574b37a3e87046000ff8447df84dce4.

Hannsek, 2023-02-27:
[image]

Hannsek, 2023-03-20:
cc @b-irving

Hannsek, 2023-03-20:
So, can Suite actually send some messages to Trezor that CJ has been stoped or resumed? If yes, I suppose the text `Coinjoin running` can remain, no?

hynek-jina, 2023-03-21:
Btw. we plan to replace `pause` and `cancel` with single option called `stop`
https://github.com/trezor/trezor-suite/issues/7871

It probably doesn't change much.. Because technically there will still be the same functionality in the background.. But from user perspective coinjoin will be stopped. So the `Coinjoin running` would be confusing in that situation. But we can stick to `pause` if that helps.

andrewkozlik, 2023-03-21:
> Btw. we plan to replace `pause` and `cancel` with single option called `stop` [trezor/trezor-suite#7871](https://github.com/trezor/trezor-suite/issues/7871)

I am not concerned about whether it will be called "pause" or "stop", but now that I thing about it, we might want to keep the "cancel" button somewhere in case the user actually wants to cancel the authorization on Trezor. They will be able to do cancel implicitly by manipulating the settings, but that's not obvious.

hynek-jina, 2023-03-21:
What about cancel after some timeout? 🤔

Hannsek, 2023-03-21:
so Coinjoin authorized is beter?
[image]