trezor/trezor-firmware #2836: Device tests in CI are green even it they failed

grdddj, 2023-02-17:
Happened at least two times, for example https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/3791042448 in https://github.com/trezor/trezor-firmware/pull/2835

matejcik, 2023-02-17:
here is a visible log output: https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/3791480246
the make target fails, but the job as a whole succeeds. is it cached-nix-shell swallowing the return value? is it `ts`?

matejcik, 2023-02-17:
well this is fun
```
$ nix-shell --run "false" ; echo $?
1
$ nix-shell --run "false | ts -s" ; echo $?
1
$ cached-nix-shell --run "false" ; echo $?
1
$ cached-nix-shell --run "false | ts -s" ; echo $?
0
```
nix-shell seems to have pipefail set (`set -o pipefail`). 
cached-nix-shell does not.
we need it.

* we can put it in front of every command where it matters: `set -o pipefail; make test_emu | ts -s`
* we can figure out how to make cached-nix-shell do it automatically -- maybe some sort of bash profile??
* we can revert to nix-shell

mmilata, 2023-02-18:
Opened xzfc/cached-nix-shell#28. I'm slightly in favor of reverting to nix-shell since we added a layer of workarounds for this and the speedup is not enough to be measurable.

matejcik, 2023-02-20:
yeah, probably the least painful route. can you please do it?

prusnak, 2023-02-20:
Yeah, I also vote for the feature revert, but maybe let's keep the `NIX-SHELL` variable instead of reverting commits?

PR #2839 (2023-02-21): Revert cached-nix-shell
Fixes #2836.
closed 2023-02-21 by matejcik