trezor/trezor-firmware #2849: UI2: style update

Hannsek, 2023-02-27:
**Design:** https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=6990%3A51701&t=GE04kwfXWS6BZWAD-4

**Spec:**

- [ ] layout
 - margins will be now 6px from the edge
 - spaces between texts - 0px 
[image]
 - spaces between "sections" - 6px
 - text centered within sections
[image]
 - dynamic width for content 
[image]
 - dynamic height for headers with/without icon: with: 228x44, without: 228x20
[image]

- [ ] buttons
 - 2 buttons area: 228x50 & 228x100
[image]
[image]56
 - small button: 56x50 
[image]
 - big button: 166x50 
[image]
 - upper button: 228x64
[image] 
 - single button: 228x50
[image]
 - middle sized button: 111x50
[image]

- [ ] pagination
 - [image]

- [ ] keyboard
 - [image]

- [ ] notifications: 228x36 
[image]

- [ ] colours: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=6837%3A52769&t=GE04kwfXWS6BZWAD-4
- [ ] fonts: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=6990%3A52090&t=GE04kwfXWS6BZWAD-4
 - Headlines and buttons: 
[image]
 - Subheadline: 
[image]
 - Body: 
[image]
 - Mono: 
[image]

- [ ] icons: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=6990%3A52122&t=GE04kwfXWS6BZWAD-4

mmilata, 2023-03-04:
Also: https://github.com/trezor/trezor-firmware/pull/2843#issuecomment-1453056584

Hannsek, 2023-03-06:
Also, the "intensity" homescreen blurring has been edited.  See: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=7079%3A54925&t=mB1OgFwgcx9jG4K2-4

PR #2876 (2023-03-23): TT UI: style update
Fixes #2849, fixes #2734 and possibly #2657.

[image] [image] [image] [image] [image] [image] [image] [image] ![fc7c6eb03347115fec4de5f255f16eb580d2b12f60aab3ec01506d0cd2289bb6](https://user-images.
closed 2023-03-23 by mmilata