trezor/trezor-firmware #2863: Small Taproot QR code on T1

andrewkozlik, 2023-03-06:
The Taproot QR code in `GetAddress` on T1 is very small. Can we make it bigger? The problem is that the code is 33x33 pixels, whereas the available area is 64x64. We could double the size and then shave off one line of pixels on each side (top, bottom, left, right). Or increase the area to, say, 74x64 and shave off only one row of pixels on the top and bottom.

Before:
[image]

After 1:
[image]

After 2:
[image]

The same problem might also be present for P2WSH, i.e. SegWit v0 multisig.

Hannsek, 2023-03-06:
There were discussions about this here https://github.com/trezor/trezor-firmware/issues/2638 and https://github.com/trezor/trezor-firmware/issues/2190.

I believe it is the same issue, no?

matejcik, 2023-03-06:
yeah, we tried the cutoff but apparently there have been problems with scannability.

prusnak, 2023-03-06:
Maybe we should drop the QR code from T1 completely?

I think it's a gimmick and no one uses it, since it is unusable (unlike TT where QR code is much bigger).

prusnak, 2023-03-06:
> Most of the users can scan it comfortably.

Even the smallest Taproot one?

Hannsek, 2023-03-06:
Yes, but it depends on the cameras.

closed 2023-03-06 by matejcik