trezor/trezor-firmware #2865: Edit output from emulator to match colours in UI2

Hannsek, 2023-03-07:
Gamma correction would be needed.

This can be done after #2835.

grdddj, 2023-03-20:
Proof of concept done in https://github.com/trezor/trezor-firmware/commit/0ccf916c3744a1e0f86ba9fdde15adc566aafea0
- building on top of https://github.com/trezor/trezor-firmware/pull/2876 to have the latest designs
- increasing the brightness of colors in the emulator
- was trying to do it with gamma correction, but was not successful
- there are a lot of ways/algorithms how to increase the color brightness and there is hardly one that will work perfectly for all colors (from almost-black to almost-white)
  - currently, I am using this way
    - the maximum values of R, G and B are 31, 63 and 31 respectively (based on RGB565)
    - take the difference between current value and its maximum value, if R is 15, the diff is 31 - 15 = 16
    - multiply this difference by some percentage `INCREASE_BRIGHTNESS_PERCENT` in code, defaulting at `25`, getting 16*0.25 = 4
    - adding this to the current value, ending up at 15 + 4 = 19
- by pressing left and right arrows when in the emulator, the  `INCREASE_BRIGHTNESS_PERCENT` is getting decreased/increased by one percentage point (and printed into the terminal), so we can experiment with its various values (personally I liked the range between 20-25) - @Hannsek it it possible to play with it on `grdddj/emu_colors` branch

The current screens are visible at https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3967679098/artifacts/test_ui_report/all_unique_screens.html - compared to older ones in https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3943227735/artifacts/test_ui_report/all_unique_screens.html

Other possibilities
- have some other "brightness-increasing" algorithm
- hard-coding the color transition for the emulator (there are not so many main colors actually)

prusnak, 2023-03-20:
> * was trying to do it with gamma correction, but was not successful

What does this mean? That you were not able to do it or that you did it but the results were not great?

grdddj, 2023-03-20:
> > * was trying to do it with gamma correction, but was not successful
> 
> What does this mean? That you were not able to do it or that you did it but the results were not great?

I managed "some" correction, but the results looked either very badly, or they made the screen colors darker (and we need to make them brighter to match hardware device). I got inspired by the code from @matejcik on Slack - in this thread - https://satoshilabs.slack.com/archives/CL1D61PQF/p1679325543818369?thread_ts=1677849553.033139&cid=CL1D61PQF

There could be some issues with that approach, I could have made a mistake implementing it, or it simply does not work in our case

grdddj, 2023-03-20:
> There could be some issues with that approach, I could have made a mistake implementing it, or it simply does not work in our case

So it was the combination of a) and b) - I copy-pasted it without changing the maximum RGB values from `255` into `31` and `63`

After changing that in https://github.com/trezor/trezor-firmware/commit/6757108bafdfdfa66441d8be0865e8a1dfc7c0d5 it seems to be working fine, and the `gamma` value of `0.55` produces good results. It is still configurable so we can find the best value

Thanks @prusnak, without your question I would completely abandon that approach :)

prusnak, 2023-03-20:
> Thanks @prusnak, without your question I would completely abandon that approach :)

Great, meanwhile I did the experiment with the gamma locally and my change looks exactly like yours - with small nitpick - I pushed that nitpick to `display-gamma-wip` branch - see 6af5ad6a019a6fa1810b330bd39bb06751d111e3 - feel free to incorporate it into your branch. Otherwise great job!

grdddj, 2023-03-20:
> > Thanks @prusnak, without your question I would completely abandon that approach :)
> 
> Great, meanwhile I did the experiment with the gamma locally and my change looks exactly like yours - with small nitpick - I pushed that nitpick to `display-gamma-wip` branch - see [6af5ad6](https://github.com/trezor/trezor-firmware/commit/6af5ad6a019a6fa1810b330bd39bb06751d111e3) - feel free to incorporate it into your branch. Otherwise great job!

Nice changes - I made them part of the PR - https://github.com/trezor/trezor-firmware/pull/2894

closed 2023-03-23 by grdddj

matejcik, 2023-03-23:
ooooohhhh now I remember why I didn't want to run the correction in display_pixeldata:

the correction should happen on **screen output only** so that it does not touch the raw screen(shot) bytes.

IOW, you would copy and gamma-correct a separate buffer before blitting it to the SDL texture, but leave the original one untouched. screenshots for fixtures would not get corrected.

the drawback is that the screenshots in UI reports would be uncorrected and thus unusable for designers / documentation / etc.
would that be a problem @Hannsek ? if screenshots taken from the emulator directly are fine

Hannsek, 2023-03-23:
It would not. We currently don't use UI reports for any design/documentation. However, what advantages does this approach bring besides no need for editing UI tests?

matejcik, 2023-03-23:
> However, what advantages does this approach bring besides no need for editing UI tests?

it's cleaner e.g. in case we ever introduce reading back pixeldata from the display.
also the gamma can currently be interactively configured, so you can in theory change gamma mid-stream and get screenshots to come out differently.

> we set `SDL_VIDEODRIVER = dummy` in this case, we could somehow read it within emulator code

sure, that's not a problem. one thing, we'd need to set a variable at start instead of reading environment every time we try to blit to screen.
but i don't expect this to have a noticeable effect, the display is small and the C code is going to be very fast.

grdddj, 2023-03-23:
> it's cleaner e.g. in case we ever introduce reading back pixeldata from the display. also the gamma can currently be interactively configured, so you can in theory change gamma mid-stream and get [...]

Changes done in https://github.com/trezor/trezor-firmware/pull/2903