trezor/trezor-firmware #2872: Act on some findings from coverage report

grdddj, 2023-03-09:
Comment - https://github.com/trezor/trezor-firmware/pull/2857#issuecomment-1460572425 - covers some possible improvements we can do
- new tests
- possible bugs

matejcik, 2023-03-10:
copied over to here so that we can have an easier discussion

---

Current coverage report: https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/index.html

Some other things I noticed when going through it:
- function definitions with decorators (even `@staticmethod` or `@classmethod`) are marked untested - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_storage_sd_salt_py.html
or 
https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_sign_tx_tx_weight_py.html#n52
- strange things are sometimes happening with `utils.BITCOIN_ONLY` - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_sign_tx___init___py.html#n7
- probably a majority of currently "untested" lines are raises - extreme example is https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_nem_validators_py.html
- I am surprised that two cases are untested in `BTC:SignatureVerifier` - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_verification_py.html#n80
- `parse_input_script_multisig` also looks like an important function - actually included in the untested branch above -  https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_scripts_py.html#n500
- `set_flags` looks like an important functionality, untested - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_storage_device_py.html#n216
- can we simulate the `usbcheck`? - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_trezor_ui_layouts_tt_v2_homescreen_py.html#n71
- `coininfo.py` - another dict-like if-if chain - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_common_coininfo_py.html
- this `trezor/ui/__init__.py` looks strange - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_trezor_ui___init___py.html
- this is also strange, the `_clear_tx_request` looks like not being called - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_sign_tx_helpers_py.html#n366
- this could even be a bug - there is no `raise` in front of `DataError` - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_sign_tx_payment_request_py.html#n48
- some animation code is untested - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_bitcoin_sign_tx_progress_py.html#n125
- do not we test this in storage tests? - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_storage___init___py.html
- `read_compact_size` would be worth at least a unit-test - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_common_readers_py.html#n7
- we might have a test cancelling the abortion from recovery homepage - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_management_recovery_device_layout_py.html#n146
- some of these errors are really never instantiated, could they be deleted not to take flash space? They take `100 bytes` each https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_trezor_wire_errors_py.html
- sometimes the coverage does not make sense, there are green lines right below red ones - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_common_keychain_py.html#n65
- looks like we do not have tests for changing rotation to some degrees - probably worth doing it - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_management_apply_settings_py.html#n191
- maybe worth a quick test (same with wipe-code) - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_apps_management_change_pin_py.html#n99
- module docstrings are being flagged - https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3896730236/artifacts/core/htmlcov/src_trezor_crypto_bech32_py.html#n21

Could we also somehow combine `unit tests` into it? It might test some of the edge-cases which raise a lot

These patterns could be also probably ignored:
```
pass
assert False
raise RuntimeError
raise NotImplementedError
def mem_dump
def __repr__(self)
```

It looks s though `click tests` were not included in the result - as `.coverage.test_click` is not included in the printout - https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/3896730236 - and it really looks like the `core/.coverage.*` is not specified in `paths` of `core click test`. Will include it.

I take back what I said about the `TR` model, when both models will be part of our CI, we will create .coverage files for both models and combine them, so nothing should be missing. (The .coverage files will need different names, however.) When running just one model locally, though, it will currently miss the other model.

matejcik, 2023-03-10:
> * strange things are sometimes happening with `utils.BITCOIN_ONLY`

that's probably because `if BITCOIN_ONLY` converts to `if True` and the bytecode compiler statically removes the check. the correct solution is to ignore `if (not)? utils.BITCOIN_ONLY` pattern

> * probably a majority of currently "untested" lines are raises

i think there's no way around that unless we want to go insane :( except perhaps if we got a fuzzer to make the coverage for us

> * I am surprised that two cases are untested in `BTC:SignatureVerifier`

are they not tested even in unit tests?
for that matter, do we have coverage for unit tests? i suspect not

> * this `trezor/ui/__init__.py` looks strange

i suspect most of it is now (or very soon will be) dead code :woman_shrugging: 

> * this is also strange, the `_clear_tx_request` looks like not being called

it looks weird but I think it's correct: the txfinish request is the last message in the signing flow and indicates to the caller that it should end the generator.
maybe we could stick a RuntimeError in place of the clearing
but in general i'd like to get rid of this generator-based weirdness in bitcoin signing Soon(tm), we already have 90 % of the necessary prerequisites in place

> * this could even be a bug - there is no `raise` in front of `DataError`

absolutely a bug. I'd love to get a lint for this thing, but can't find one. it _seems_ this would be not difficult to test for - even if you don't know that a symbol represents an exception, you can know relatively well that it's a constructor and its result is unused? 
but [I found nothing](https://stackoverflow.com/questions/75694674/lint-to-check-for-a-missing-raise-keyword)

> * do not we test this in storage tests?

upgrade-tests should be covering this, but they presumably also don't generate coverage?

matejcik, 2023-03-10:
> Could we also somehow combine `unit tests` into it? It might test some of the edge-cases which raise a lot

probably! unit tests execute directly in the micropython interpreter, maybe shoving the coverage wrapper in front would Just Work(tm)?