trezor/trezor-firmware #2878: Taproot address generated by the

octopusengine, 2023-03-12:
I'm trying to use the Python library to generate addresses (in my test application) and I'm not sure about taproot.

In previous versions I used:
```
44'/0'/0'/0/0 default bitcoin SPENDADDRESS address 0: 16MkCp4e3sSCc...8ZKJ7BBZ9dHhV - LEGACY ACCOUNTS ok
49'/0'/0'/0/0 bitcoin SPENDP2SHWITNESS address:       3DSrzWmuP31W9...eEMJWGEqrkxFj - LEGACY SEGWIT ACCOUNTS ok
84'/0'/0'/0/0 bitcoin SPENDWITNESS address:           bc1qdkknh0drn...fmxzwsrp80cnp - DEFAULT ACCOUNTS ok
```
And for taprtoot I'm trying path 86/... is that correct? The address does not match the Trezor generated
```
86'/0'/0'/0/0 bitcoin SPENDTAPROOT address:           bc1pq37dqchmun...hnlgdwrxwssuxkzw - 0? 
```

Is it possible that it starts from 1 and not from 0 as in the previous cases? I am missing the path display in the app including the index, is that option available somewhere?

octopusengine, 2023-03-12:
It really looks like the first address in the Vault was not 0. For the 86'/0'/0/1 path it is already correct (and the addresses match of course) , but I missed checking in Suite. And also the ability to derive the address with some higher index.

prusnak, 2023-03-12:
> 'm trying to use the Python library to generate addresses

Which library are you using? It sounds like there is something weird going on with it.

Can you try the library on test vectors from SLIP-0014? https://github.com/satoshilabs/slips/blob/master/slip-0014.md

octopusengine, 2023-03-12:
I got it like this:  
```
...
from trezorlib.tools import parse_path
from trezorlib import btc, misc, ui
from trezorlib.messages import InputScriptType
```

And I use this
```
...
address = btc.get_address(client, "Bitcoin", bip32_path, show_display=show_disp, script_type=btc_script_type) 
```

The library seems to be working properly. Only Trezor started at "index" 1 and that confused me.
So it's more of a Suite note... 
And whether there is any way to see the full paths to the addresses as it was before.

prusnak, 2023-03-12:
> Only Trezor started at "index" 1 and that confused me.

Are you sure there are no unused addresses on this account? Suite only shows unused addresses where you are looking.

octopusengine, 2023-03-12:
I really only have one/single transaction in there, which I expected to be at position 0 - it confused me that it was at 1 (only now verified)
But it's missing at 0. (Too bad I don't see it anywhere in the Suite) 
Is it possible that even if I don't send anything to the generated address the next derived is incremented?
(But I'm not aware of that happening.)

matejcik, 2023-03-13:
looks like whatever the issue is, it is unrelated to firmware

closed 2023-03-13 by matejcik