trezor/trezor-firmware #2895: UI2: popup bubbles

Hannsek, 2023-03-21:
Popup bubbles, now only for PIN keyboard.

[image]

design in figma: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=7523-55110&t=l6cyhdKZv0HsAB5F-4

PoC: https://satoshilabs.slack.com/archives/C02BY51RD8R/p1673610534314059

TychoVrahe, 2023-03-24:
- not sure this falls into "quite simple" scope of shape. this may not be possible
- the design should specify how the bubbels should be positioned reletive to the position touched on the screen
- it should also be specified how the bubbles should behave when on the edge of the board

Hannsek, 2023-03-24:
The bottom part of the shape above is the actual button. So when you click on the button, it will be “overdrawn” by this.

closed 2024-05-07 by Hannsek