trezor/trezor-firmware #2909: Transaction signed and broadcasted but unable to locate TxnHash

iw4p, 2023-03-26:
**Describe the bug**
I tried to create and sign a transaction via trezorlib (script) and trezorctl (CLI). Both were successful but when I tried to broadcast them with [etherscan](https://etherscan.io/pushTx) and the `w3.eth.send_raw_transaction` method I got broadcasted response but nothing happened.

**Firmware version and revision**
1.11.2

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite: Fine, Trezorctl: Not Fine, [This script](https://github.com/brianddk/reddit/blob/master/python/erc20_offl_txn.py): Not fine
 - OS: macOS
 - Browser: Brave
 - Version: 1.11.2

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/brianddk/reddit/blob/master/python/erc20_offl_txn.py
2. Change ERC20 token address and destination and amount
3. Run script
4. See Transaction broadcasted: https://etherscan.io/tx/HEXVALUEHERE
5. Try again and see already known.
6. I also tried to broadcast a transaction with etherscan. same problem.

**Expected behavior**
when I open etherscan link I need to see the broadcasted transaction. but it says Sorry, We are unable to locate this TxnHash

**Additional context**
I'm trying to send the ERC20 token to another address.

Hannsek, 2023-03-26:
It looks more like Connect issue. cc @mroz22

prusnak, 2023-03-26:
> It looks more like Connect issue. cc @mroz22

Not really, Connect is not being used here.

matejcik, 2023-03-27:
the script you are linking to does not actually broadcast the transaction.
neither does your `trezorctl` invocation, unless you add `--publish` option.

you're saying you failed to broadcast via etherscan, but that could be a problem on their end.

also, the tx hex that you are getting out of trezorctl / custom script is _something other_ than the TxnHash. if you paste it into etherscan, chances are it can't find it.
the broadcast page should give you the txnhash, or you'll get it from `trezorctl ethereum sign-tx (...) --publish`

iw4p, 2023-03-27:
> the script you are linking to does not actually broadcast the transaction. neither does your `trezorctl` invocation, unless you add `--publish` option.
>
> you're saying you failed to broadcast via [...]

I tried to push it with https://etherscan.io/pushTx
The problem was wrong nonce. Sorry, and thanks for your support!

closed 2023-03-27 by Hannsek