trezor/trezor-firmware #2910: Bootloader UI: RSOD

Hannsek, 2023-03-27:
[image]

https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=7657%3A55243&t=Mmm1eRg2tu68cdMZ-4

Hannsek, 2023-03-27:
Please change these screens as well:
[image]

[image]

matejcik, 2023-03-29:
also changing the initial installation screen, hope that's ok
[image]

Hannsek, 2023-03-29:
Is this done on the top of master? Because I see smaller font than it should be. Otherwise ok.

Hannsek, 2023-03-29:
Which colour did you choose for the thin line?

matejcik, 2023-03-29:
> Is this done on the top of master? Because I see smaller font than it should be. Otherwise ok.

the font is smaller because it's the same one as the "RESTARTING". in bootloader we only have the normal non-bold font. I can put in that one.

> Which colour did you choose for the thin line?

some kind of gray, same as was the button color previously

Hannsek, 2023-03-29:
Please use TT/Body (TT Hoves, Regular, 21)

matejcik, 2023-03-30:
[image]

Hannsek, 2023-03-30:
The `restartinng in 3` looks like it is not centered between the line and the bottom of the screen.

matejcik, 2023-03-31:
[image] [image] [image] [image] [image] [image] [image]

In case of errors, I can tweak _some_ strings:
* I can globally change "FATAL ERROR" to "INTERNAL ERROR"
* I can always change contents of the middle line
* In case of "USB ERROR", I can also separately modify the title too.

please let me know what the wording should be. we still need to remove the full-stops at end of "installed successfully."

> The `restartinng in 3` looks like it is not centered

that's an artifact of the screenshot: the "virtual screen" does not extend all the way to the bottom edge of the screen cutout.
even if the real device is the same, it would be annoying to deal with it: we would need to make the black-background text move a couple pixels down to compensate, but leave alone all those with non-black background. I would prefer not doing that in the bootloader.

matejcik, 2023-04-03:
> invalid firmware signature

there are actually several "reasons" for ignoring the firmware that I think is too much:

* Bootloader downgraded

keep this one but change to "Bootloader downgrade protection" ? current users will not see it anyway.

* invalid vendor header
* invalid vendor header signature
* invalid firmware header
* invalid firmware signature
* invalid firmware hash

=> all of those can be something like "firmware is corrupted", I have a hard time imagining a scenario where the distinction is important. In all cases the resolution steps are (1) try to reinstall firmware, several times, and (2) throw out the device and get a new one.

In the new-style user friendly error screens, I would like "FIRMWARE IS CORRUPTED" (or similar) to be the header, and the reason code be the smaller text. For now, "internal error" / "firmware corrupted" seems fine.

* unauthorized vendor keys 

This is a very special case that nobody will ever see, so i'd leave that alone, except maybe capitalize the first word

* wrong firmware model

This one seems relevant, although in theory it should be impossible to encounter.

Hannsek, 2023-04-03:
[image]
Should start with big "I"

commit 5361af38 (2023-03-31): fixup! feat(core/bootloader): new style for result screens (fixes #2910)
commit cb6f9311 (2023-04-03): feat(core/bootloader): new style for result screens (fixes #2910)
commit 262a589c (2023-04-03): feat(core/bootloader): new style for result screens (fixes #2910)
commit 839a9409 (2023-04-03): feat(core/bootloader): new style for result screens (fixes #2910)
closed 2023-04-03 by matejcik