trezor/trezor-firmware #2930: Do not display numbers in header when there is only one address and amount.

Hannsek, 2023-04-03:
[image]

In the send flow, currently, there are `RECIPIENT #1` and `AMOUNT #1` even if user sends funds to only one address.

Desired outcome:

When user sends funds to only one address, do not display `#X`.

matejcik, 2023-04-03:
we can't easily do this because we don't know how many recipients will be needed

(in the happy-path tx with two outputs, one of which is change, we get `outputs_count=2`. we cannot know in advance whether the second output will really be change or whether it is, e.g., malformed and we will be showing it.)

we could either pre-stream all outputs and count the number of displayed ones (which slows down signing)
or let Suite tell us to hide the counters (might be the best -- the counters are a UX feature, not a security feature)
or do something more involved (like if the "hide_counters" flag is set, and we encounter more than one non-change output, cancel signing because it meant that the host had lied to us)
or just hide "#1" and if there are more, show "RECIPIENT #2" for the second one

Hannsek, 2023-04-03:
What to you mean by malformed here?

> we could either pre-stream all outputs and count the number of displayed ones (which slows down signing)
Slows how much?

matejcik, 2023-04-03:
like, there are change-output details but they do not match the transaction, e.g., the change is going to another account.
or for that matter it might not be a change output at all but another recipient

> Slows how much?

Turns out not that much. The slowdown depends on the number of _outputs_, which is two in the usual case and maybe ~10 for big senders. Unless someone is doying payroll to hundreds of employees, in which case the total slowdown could be on the order of tens of seconds before they can start confirming.
(and even then, we could cut short when we figure out that multiple recipients are needed)
The only issue might be coinjoin -- where we might be able to skip that step, at the cost of a little complexity.

---

If we got Suite to cooperate and send the change output as the first of the two, we could employ a trick: "if this is the last output and it's the first shown on display, then hide the number". That seems most elegant, and would be very little work on firmware side too.

Hannsek, 2023-04-03:
Ok, Suite can cooperate for sure. 👍🏻