trezor/trezor-firmware #2938: Change text on the button

Hannsek, 2023-04-06:
While setting locktime, on the confirmation button, there is currently `NEXT`. 

Please change it to `CONFIRM`.

mmilata, 2023-04-11:
Wanna get rid of NEXT everywhere? These screens would be affected: https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4092339281/artifacts/test_ui_report/differing_screens.html

Hannsek, 2023-04-11:
Probably yes. But not replace it with `confirm` everywhere.

For example here:

[image] 

and here: 
[image]

`Continue` would fit better here.

Also, how much complexity would be to add custom text in different flows?
E.g. here: 
[image]

`Decrypt` instead of `Confirm`

Hannsek, 2023-04-11:
Yes, it should. Something like this is feasable?

[image]

Also, this is how signing unknown token (not in eth definition) would look like?
[image]

prusnak, 2023-04-11:
Does the locktime (YYYY-MM-DD HH:MM:SS)  fit on the display with the monospace font? If yes, it would be nice to change it.

Hannsek, 2023-04-11:
It unfortunately won't fit by few pixels.
However, we can do something like this:
[image]

PR #2950 (2023-04-12): TT: button text and warning style update
Fixes #2938. No more `NEXT` buttons. Some screens that had _Warning_ in the title were converted to use the warning layout with icon. Changed button style on the warning layout.

[UI diff](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4095806738/artifacts/test_ui_report/index.html) | [differing screens](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4095806738/artifacts/test_ui_report/differing_screens.html)

[image] [image] [image] [image] [image] [image]
closed 2023-04-12 by mmilata