trezor/trezor-firmware #2949: Investigate coinjoin signing slowdown

mmilata, 2023-04-11:
The `test_authorize_coinjoin.py::test_sign_tx_large` testcase often fails in hardware tests because of a duration check:
```
04:34:40 >       assert delay <= max_expected_delay
04:34:40 E       assert 61.11028814315796 <= 60
```
https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/4082717229

The testcase was introduced fairly recently - check if the firmware from that time performs better. If there is a significant difference, use git-bisect to try to find which change caused it.

andrewkozlik, 2023-09-18:
FTR, we had a discussion about this problem [here](https://satoshilabs.slack.com/archives/CL1D61PQF/p1688402815622609?thread_ts=1688399906.699769&cid=CL1D61PQF). @onvej-sl bisected the commits and found that the increase in time is caused by updating micropython: https://github.com/trezor/trezor-firmware/commit/3fb3f108ccf49df8ffa8f3ff792f0cc7fec467e6.

andrewkozlik, 2023-09-18:
Actually https://github.com/trezor/trezor-firmware/commit/3fb3f108ccf49df8ffa8f3ff792f0cc7fec467e6 was committed after this issue was opened. The micropython update (and possibly other commits between April and July 2023) seem to have introduced a slowdown of +15 seconds. The slowdown that was observed in the OP may be due to the fact that tests take longer in the CI, probably because of task switching when multiple jobs are being run. IIRC, we also found that the signing takes a significantly different time on different machines.

closed 2025-02-05 by romanz