trezor/trezor-firmware #2952: Add ability to permanently "become" a sub-key, forgetting all parent/sibling keys (analogous to "drop root priveledges", "chroot jail", etc)

SecurityIsHard69420666, 2023-04-13:
My understanding of HD Wallets is primitive but I'm sure one of you can take the idea and run with it.

Motivation:
1) I consider it to be dangerous to keep your Trezor initialized with your seed phrase, even if protected by a secondary security feature like a PIN or SD Card. Cameras/eyeballs can steal your PIN, Grab 'n Run to steal the Trezor itself. It should be noted that a really strong passphrase *is* good security but only because it is similar to *not* having your Trezor [fully] initialized.
2) I don't want *more* seed phrases and/or passphrases to manage. Since HD Wallets can generate endless keys, you really shouldn't ever need more seed or pass phrases.
3) I want to use things like FIDO2, ssh, gpg, and Sign-In With Ethereum with a key that doesn't have a significant amount of Eth in it (maybe just a little bit of money for day to day spending). The very act of using my Trezor daily (esp in public) opens up a huge security vector.

You can almost achieve what I want by doing (3) activities on the "seed phrase" (no passphrase) level and then keeping large amounts of crypto behind a really strong passphrase, but that violates (2).

Proposal:
It should be possible to take one of the child sub-keys/DPs and to set/overwrite it on the Trezor to be the new "root" private key, allowing only access to that sub-tree and permanently/securely (until re-initializing with seed phrase) removing the ability to interact with the parent/sibling trees. Obviously it would become impossible to 'verify' the seed phrase at this point but that's expected behavior. If the way HD Wallets works prohibits doing this then surely there's some workaround that can be devised to get similar behavior, even if none of the new keys were in the old tree. For example take the "root" key that's in memory when using a "seed phrase + passphrase combo" and save/overwrite the "seed phrase only (no passphrase)" root key that's stored in Trezor's nonvolatile memory. The passphrase need only be something like "0" or "a" (or something deterministic/not-user-chosen by Trezor itself) and it would still give you the security properties desired.

How it might look:
`$ trezorctl --become-subkey "m/44'/60'/0'/0"`
OR
`$ trezorctl --overwrite-root-key-with-passphrase-key`
OR
`$ trezorctl --become-subtree 0 #we could get to subtree "0" by setting "0" as a passphrase, or there could be some other deterministic algorithm used`

And it'd output a "WARNING: You won't be able to access parent/sibling keys until you re-initialize, also 'verifying' the seed phrase won't work anymore"

Conclusion:
With just 1 seed phrase to manage, you can use your Trezor (as in (3) above) knowing that even if your Trezor+PIN are stolen, the keys that are storing your *larger* amounts of crypto are still safe.

Hannsek, 2023-04-13:
So, generally, you want one Trezor device per passphrase, so you don't need to remember the passphrase. Right?

prusnak, 2023-04-13:
> So, generally, you want one Trezor device per passphrase, so you don't need to remember the passphrase. Right?

Not really. This is more like let's initialize 3 devices with the same seed and then lock one to Bitcoin first account (`m/44'/0'/0'`), other one to Ethereum second account (`m/44'/60'/0'`) and the last one only to use with FIDO2 for example. So if you try to use the Ethereum device with Bitcoin, it wouldn't work.

While I see some appeal of the proposal, I feel this is a very niche use-case which does not really justify the amount of engineering required. Also the existence of such device mechanism would cause lot of confusion in third parties, because we would need to devise a way how to indicate such lock exists and a third party would need to reflect that.

Hannsek, 2023-04-13:
I don't think we will implement this anytime soon. I am closing this and will reopen once it is actual.

closed 2023-04-13 by Hannsek

SecurityIsHard69420666, 2023-04-18:
@Hannsek if I submit a PR implementing this, will it be accepted? it can be hidden behind some --experimental flag or just not mentioned in the --help, I don't care.

I'd call it --make-passphrase-permanent for now, but we could talk about the name itself in the PR.

Hannsek, 2023-04-18:
Not right now. We don't have enough capacity to review your to be PR at the moment. 

However, you can build your own fw with this feature included and so you can use Trezor _for buying your daily cup of coffee._