trezor/trezor-firmware #2957: Coinjoin screens for T2B1

Hannsek, 2023-04-13:
**Design:** To be specified

grdddj, 2023-04-13:
Draft without loader done in [e1252e9](https://github.com/trezor/trezor-firmware/pull/2610/commits/e1252e9ee9f9a2415f29a4e47d15b80ad34f1f1b):

[image]

Hannsek, 2023-04-25:
[image]

Hannsek, 2023-06-01:
Is this done?

grdddj, 2023-06-01:
> Is this done?

No, the loaders are missing there

Hannsek, 2023-06-07:
Small changes: Font in header, wording

https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?type=design&node-id=8825-68716&t=kyXA4olzpDvXBNZr-4

grdddj, 2023-07-17:
Merged in https://github.com/trezor/trezor-firmware/pull/3135

closed 2023-07-17 by grdddj