trezor/trezor-firmware #2963: MetaMask Message Signature: Error: Firmware error

parmesh-shiroya, 2023-04-16:
Trezoe Model T is throwing `MetaMask Message Signature: Error: Firmware error` msg when trying to signTypeData.

Firmware version : 2.5.3

**To Reproduce**
Steps to reproduce the behavior:
You can reproduce the issue by going to opensea.io and try to bulk list with more than 3 items. (It works with 2 items)
domain :
```{
    "name": "Seaport",
    "version": "1.4",
    "chainId": 1,
    "verifyingContract": "0x00000000000001ad428e4906ae43d8f9852d0dd6"
}```

types :
```
{
    "BulkOrder": [
        {
            "name": "tree",
            "type": "OrderComponents[2][2]"
        }
    ],
    "OrderComponents": [
        {
            "name": "offerer",
            "type": "address"
        },
        {
            "name": "zone",
            "type": "address"
        },
        {
            "name": "offer",
            "type": "OfferItem[]"
        },
        {
            "name": "consideration",
            "type": "ConsiderationItem[]"
        },
        {
            "name": "orderType",
            "type": "uint8"
        },
        {
            "name": "startTime",
            "type": "uint256"
        },
        {
            "name": "endTime",
            "type": "uint256"
        },
        {
            "name": "zoneHash",
            "type": "bytes32"
        },
        {
            "name": "salt",
            "type": "uint256"
        },
        {
            "name": "conduitKey",
            "type": "bytes32"
        },
        {
            "name": "counter",
            "type": "uint256"
        }
    ],
[... 191 lines cut ...]
                "orderType": 0,
                "zone": "0x004C00500000aD104D7DBd00e3ae0A5C00560C00",
                "zoneHash": "0x0000000000000000000000000000000000000000000000000000000000000000",
                "salt": "51951570786726798460324975021501917861654789585098516727729426175207949446544",
                "conduitKey": "0x0000007b02230091a7ed01230072f7006a004d60a8d4e71d599b8104250f0000",
                "totalOriginalConsiderationItems": 3,
                "counter": "0"
            },
            {
                "offerer": "0x0000000000000000000000000000000000000000",
                "zone": "0x0000000000000000000000000000000000000000",
                "offer": [],
                "consideration": [],
                "orderType": 0,
                "startTime": 0,
                "endTime": 0,
                "zoneHash": "0x0000000000000000000000000000000000000000000000000000000000000000",
                "salt": 0,
                "conduitKey": "0x0000000000000000000000000000000000000000000000000000000000000000",
                "counter": 0
            }
        ]
    ]
}
```

matejcik, 2023-04-24:
> `"type":"OrderComponents[2][2]"`

arrays of arrays are currently not supported on Trezor

matejcik, 2023-04-25:
> This works on Trezor one.

It doesn't really, Trezor One is only signing hashes that someone else has generated. Literally _anything_ will "work" on Trezor One regardless of whether it is even a valid EIP712 message.

Anyway, no ETA for this, sorry

prusnak, 2023-04-25:
> Anyway, no ETA for this, sorry

Haven't we implemented blind signing on Model T too?

matejcik, 2023-04-25:
> Haven't we implemented blind signing on Model T too?

We did a draft implementation as a stop-gap in case Metamask couldn't integrate the full featured version, but we solved that one. And we don't really want blind signing on the TT unless there is a very strong need. (in which case prioritizing the effort on this issue would be a better way to go)

parmesh-shiroya, 2023-04-29:
If nested array is not supported. It should not work for 2 item also. Right? 
For 2 items it's working. 🤔 
domainData
`{
    "name": "Seaport",
    "version": "1.4",
    "chainId": 1,
    "verifyingContract": "0x00000000000001ad428e4906ae43d8f9852d0dd6"
}`

type
`{
    "BulkOrder": [
        {
            "name": "tree",
            "type": "OrderComponents[2]"
        }
    ],
    "OrderComponents": [
        {
            "name": "offerer",
            "type": "address"
        },
        {
            "name": "zone",
            "type": "address"
        },
        {
            "name": "offer",
            "type": "OfferItem[]"
        },
        {
            "name": "consideration",
            "type": "ConsiderationItem[]"
        },
        {
            "name": "orderType",
            "type": "uint8"
        },
        {
            "name": "startTime",
            "type": "uint256"
        },
        {
            "name": "endTime",
            "type": "uint256"
        },
        {
            "name": "zoneHash",
            "type": "bytes32"
        },
        {
            "name": "salt",
            "type": "uint256"
        },
        {
            "name": "conduitKey",
            "type": "bytes32"
        },
        {
            "name": "counter",
            "type": "uint256"
        }
    ],
    "OfferItem": [
        {
            "name": "itemType",
            "type": "uint8"
        },
        {
            "name": "token",
            "type": "address"
        },
        {
            "name": "identifierOrCriteria",
            "type": "uint256"
        },
        {
            "name": "startAmount",
            "type": "uint256"
        },
        {
            "name": "endAmount",
            "type": "uint256"
        }
    ],
    "ConsiderationItem": [
        {
            "name": "itemType",
            "type": "uint8"
        },
        {
            "name": "token",
            "type": "address"
        },
        {
            "name": "identifierOrCriteria",
            "type": "uint256"
        },
        {
            "name": "startAmount",
            "type": "uint256"
        },
        {
            "name": "endAmount",
            "type": "uint256"
        },
        {
            "name": "recipient",
            "type": "address"
        }
    ]
}`

value
`{
    "tree": [
        {
            "offerer": "0x6967AC5b17e1961b80042F1333b3E4146228c6Ff",
            "offer": [
                {
                    "itemType": 2,
                    "token": "0xd07e72b00431af84ad438ca995fd9a7f0207542d",
                    "identifierOrCriteria": "2263",
                    "startAmount": "1",
                    "endAmount": "1"
                }
            ],
            "consideration": [
                {
                    "itemType": 0,
                    "token": "0x0000000000000000000000000000000000000000",
                    "identifierOrCriteria": 0,
                    "startAmount": "760000000000000000",
                    "endAmount": "760000000000000000",
                    "recipient": "0x6967AC5b17e1961b80042F1333b3E4146228c6Ff"
                },
                {
                    "itemType": 0,
                    "token": "0x0000000000000000000000000000000000000000",
                    "identifierOrCriteria": 0,
                    "startAmount": "40000000000000000",
                    "endAmount": "40000000000000000",
                    "recipient": "0xbd3af18e0b7ebb30d49b253ab00788b92604552c"
                }
            ],
            "startTime": "1682776271",
            "endTime": "1684603426",
            "orderType": 0,
            "zone": "0x004C00500000aD104D7DBd00e3ae0A5C00560C00",
            "zoneHash": "0x0000000000000000000000000000000000000000000000000000000000000000",
            "salt": "51951570786726798460324975021501917861654789585098516727718695724363952611544",
            "conduitKey": "0x0000007b02230091a7ed01230072f7006a004d60a8d4e71d599b8104250f0000",
            "totalOriginalConsiderationItems": 2,
            "counter": "0"
        },
        {
            "offerer": "0x6967AC5b17e1961b80042F1333b3E4146228c6Ff",
            "offer": [
                {
                    "itemType": 2,
                    "token": "0xd07e72b00431af84ad438ca995fd9a7f0207542d",
                    "identifierOrCriteria": "1271",
                    "startAmount": "1",
                    "endAmount": "1"
                }
            ],
            "consideration": [
                {
                    "itemType": 0,
                    "token": "0x0000000000000000000000000000000000000000",
                    "identifierOrCriteria": 0,
                    "startAmount": "760000000000000000",
                    "endAmount": "760000000000000000",
                    "recipient": "0x6967AC5b17e1961b80042F1333b3E4146228c6Ff"
                },
                {
                    "itemType": 0,
                    "token": "0x0000000000000000000000000000000000000000",
                    "identifierOrCriteria": 0,
                    "startAmount": "40000000000000000",
                    "endAmount": "40000000000000000",
                    "recipient": "0xbd3af18e0b7ebb30d49b253ab00788b92604552c"
                }
            ],
            "startTime": "1682776271",
            "endTime": "1684603426",
            "orderType": 0,
            "zone": "0x004C00500000aD104D7DBd00e3ae0A5C00560C00",
            "zoneHash": "0x0000000000000000000000000000000000000000000000000000000000000000",
            "salt": "51951570786726798460324975021501917861654789585098516727723016530330680324544",
            "conduitKey": "0x0000007b02230091a7ed01230072f7006a004d60a8d4e71d599b8104250f0000",
            "totalOriginalConsiderationItems": 2,
            "counter": "0"
        }
    ]
}`