trezor/trezor-firmware #2975: CoSi signing from command line is broken on TT

matejcik, 2023-04-21:
When introducing CoSi for TT, we changed cosi_commit + cosi_sign into a single flow.

This breaks trezorctl, because it sends Initialize before every message. This kicks TT out of the flow and it can't follow up with `CosiSign`. As a result, you get "unexpected message" every time.

Options:
* firmware-side, turn it into two separate flows again. This has _some_ advantages: logically, the CoSi protocol is two-turn, with the commit being a separate thing that could plausibly be done at a different time.
   * OTOH, the nonce is kept in short-lived cache anyway and you are technically supposed to do this online
* trezorctl side, convert the CoSi signing command so that it runs interactively and prompts the user to enter the global commitments within that one session.
   * sucks for scriptability, but then again if you're scripting the CoSi flow, just write it out in python directly
* or something something allow the cosi-sign command to proceed without `Initialize`, maybe the same way `debug send-bytes` does it, without instantiating a client at all?
  * sounds unsystematic

matejcik, 2024-02-07:
related: #3442

obrusvit, 2024-02-08:
The solution is purging CoSi from FW completely. 
Also related: #2675

PR #3551 (2024-07-01): Remove CoSi functionality from trezorctl

Resolves #2975
closed 2024-07-01 by matejcik