trezor/trezor-firmware #2978: Since the latest firmware update, why do I need to re-enter phrasepassword for every transfer by metamask?

0xshandy, 2023-04-24:
Since the latest firmware update, why do I need to re-enter phrasepassword for every transfer by metamask?

Hannsek, 2023-04-24:
Can you provide more info?

0xshandy, 2023-04-24:
> 
Before the latest firmware update, all I had to do was connect to trezor, open metamask, type in phrasepassword once, and then I could use that address to sign or transfer money until I disconnected the trezor.  But after updating the latest firmware, I found that I need to re-enter my phrasepassword every time I sign and transfer coin.

0xshandy, 2023-04-24:
> Can you provide more info?

Before the latest firmware update, all I had to do was connect to trezor, open metamask, type in phrasepassword once, and then I could use that address to sign or transfer money until I disconnected the trezor. But after updating the latest firmware, I found that I need to re-enter my phrasepassword every time I sign and transfer coin.

0xshandy, 2023-04-24:
> You have model T or model One?

model T sir

0xshandy, 2023-04-24:
> Hmm, we unfortunately cannot reproduce it. Does this happen for all coins?

Instead of using the suite client, I use the metamask plug-in on Chrome, which requires retyping phrasepassword for everything from signing to transferring money

Hannsek, 2023-04-24:
I get that, but we cannot reproduce it with Metamask. Do you use other extensions? Can you try it with some other one, like Enkrypt?

0xshandy, 2023-04-24:
> I get that, but we cannot reproduce it with Metamask. Do you use other extensions? Can you try it with some other one, like Enkrypt?

I rebooted my mac and fixed the problem, thank you

Hannsek, 2023-04-24:
Can you please still provide us with versions of Chrome browser, Metamask and Trezor Model T firmware?

0xshandy, 2023-04-24:
> Can you please still provide us with versions of Chrome browser, Metamask and Trezor Model T firmware?

Chrome 112.0.5615.137 (arm64)
MetaMask 10.28.3
ModelT firmware 2.6.0

0xshandy, 2023-04-24:
> @0xshandy Are you on macOS 13.3? Do you use Trezor Bridge with Metamask or do you use WebUSB to connect to Trezor?

13.3.1 macos I think it's Trezor Bridge with Metamask, Sir

0xshandy, 2023-04-24:
> @0xshandy Are you on macOS 13.3? Do you use Trezor Bridge with Metamask or do you use WebUSB to connect to Trezor?

I'm not sure the bridge or webusb, I on the chrome and link metamask and click transfer and then pop up a window, the url is https://connect.trezor.io/

closed 2023-04-24 by Hannsek