trezor/trezor-firmware #2990: Upgrade to micropython 1.20

prusnak, 2023-04-29:
https://github.com/micropython/micropython/releases/tag/v1.20

ISTM that the following changes might be interesting for us:
* addition of hex/fromhex methods to bytes/memoryview/bytearray (so we can drop `binascii.{,un}hexlify` usage)
* dictionary union (PEP 584)
* UTF-8 validation of all strings
* support for `__dict__` on module object
* STM32H723 support

matejcik, 2023-05-02:
> There is now an option to have the GC heap split across more than one pool of memory, although this is not yet in use by the ports.

hello CCMRAM

matejcik, 2023-05-02:
also this one is possibly interesting:

> A new port has been added, the "embed" port, which is a port of MicroPython that targets the C language and outputs a set of self-contained .c and .h files for embedding into a wider project. An example is provided to show how this works.

mmilata, 2023-10-10:
https://github.com/micropython/micropython/releases/tag/v1.21.0

prusnak, 2024-03-23:
There is now https://github.com/micropython/micropython/releases/tag/v1.22.2

prusnak, 2024-06-03:
There is now https://github.com/micropython/micropython/releases/tag/v1.23.0

prusnak, 2025-08-09:
https://github.com/micropython/micropython/releases/tag/v1.26.0