trezor/trezor-firmware #2991: RBF screens copy adjustment

Hannsek, 2023-05-02:
[image]

mmilata, 2023-05-17:
Re: first screen: we currently display  what kind of replacement is going on in the title:
* PayJoin
* Finalize transaction
* Meld transactions
* Update transaction

Do we really want to hide this information?

Hannsek, 2023-05-17:
Let's talk about this on standup.

mmilata, 2023-05-22:
For context the logic for choosing the subtitle is following. The replacement transaction flow happens when client sends a `SignTx` call with at least one input or output referring to an original input or output it is replacing.

* If the external inputs (i.e. not belonging to the walled) of the new tx form a superset of the original external inputs then `PayJoin` is displayed.
* If the tx being signed has RBF disabled show `Finalize transaction`.
* If the current inputs/outputs refer to more than one original transaction show `Meld transaction`.
* Otherwise show `Update transaction`.

Also the TXID displayed refers to the _original_ transaction so changing the title to `New transaction ID` is wrong. In case of `Meld transaction` the screen is shown multiple times, for each original transaction.

Hannsek, 2023-05-22:
> Also the TXID displayed refers to the original transaction
Are you sure about this? I believe it is the new one, but I'll try it tomorrow.

mmilata, 2023-05-22:
I didn't test it, it is my understanding of the code that handles it: https://github.com/trezor/trezor-firmware/blob/core/v2.6.0/core/src/apps/bitcoin/sign_tx/approvers.py#L260

Hannsek, 2023-05-23:
You were right, it is the old one.

PR #3029 (2023-06-02): RBF UI adjustments
Fixes #2911, fixes #2991.

* [master diff](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4345498859/artifacts/master_diff/index.html)
* [differing screens](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4345498859/artifacts/master_diff/differing_screens.html)

[image] [image] [image] [image] [image] [image] [image] ![f60e53aa7ab64bc907a8a98e8f60bcf8ff398ccf9fffd2cd38ece8c
closed 2023-06-02 by mmilata