trezor/trezor-firmware #2993: Fix off-by-one in rust geometry methods

mmilata, 2023-05-03:
https://github.com/trezor/trezor-firmware/pull/2803#discussion_r1178894504

matejcik, 2023-05-03:
after looking into it, istm this is _not_ an off-by-one at all.

instead, it's confusion between using `Point` as a _boundary_ value (this is how `Rect` uses it) and a _pixel_. 

like so:
[image]

the corner functions `bottom_left` `bottom_right` `top_right` are defined as _boundaries_ because that's how `Rect` works: the rect between 0,0 and 2,2 has size 2 in both dimensions and includes the pixels 0,0 and 1,1, but excludes pixel 2,2

this is also the right interpretation for centering, which is most of usages of the corner functions: we are looking for a point exactly between the _boundaries_.

the usage in `corner_points` cares about the corner _pixels_ and must apply correction.

the nicest solution would be to keep `Point` to always mean "boundary" (that's what `Point::center` relies on btw, otherwise it's semantically nonsense) -- and, if we deem it useful, get another type (`Pixel`?) to represent pixel coordinates.

I would slightly prefer to completely remove `bottom_left` `bottom_right` `top_right`, to reduce confusion. (`top_left` is ok because it's the same in both representations). But it would probably be good enough to add comments that they return the top-right (etc.) _boundary_.

Then `corner_points` can add a comment explaning that the offsets are for boundary corrections.

mmilata, 2023-05-03:
Hmm, I'd expect e.g. `bottom_right` to return pixel coordinates[1], but deleting these sounds alright too.

[1] does not work for empty rectangles though, better not think about negative sized-ones

matejcik, 2023-05-23:
let's "fix" this by clearly documenting current behavior. in #2610 some of the corner methods are used in the right (boundary-based) way, so using them as pixel-based seems to actually be the minority usecase