trezor/trezor-firmware #3000: Summary and info banner rendered during hold to confirm state

bosomt, 2023-05-07:
**Describe the bug**
Summary and info banner is rendered during hold to confirm state, its artifact from previous screen.
Maybe we can consider to add different banner or remove SUMMARY (i) during this phase ?

**Firmware version and revision**
 - Device: model T 2.6.0 Universal (revision 88e1f8c7a5c7615723664c64b0a25adc0c409dee)

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare transaction
2. Hold to confirm when asked by device/Suite
3. Observe screen

**Screenshots**
[image]

bosomt, 2023-05-10:
yes it happens in all hold to screens
previous screen "banner" is kept.
but only in BTC like coins send transaction scenarios it looks strange thanks to (i) that was added recently.
Not sure if we wont to "fix" it or how

Hannsek, 2023-05-10:
Ok, thanks. It is a bug. It should look like this
[image]

bosomt, 2023-10-17:
**QA OK**

**Info:**
 - Suite version: desktop 23.11.0 (40b035b2be45a4792adc9fb8373fdacff7b89bb3)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/23.11.0 Chrome/116.0.5845.188 Electron/26.2.1 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: Trezor T2T1 2.6.3 regular (revision f7aec0aeb1654268cddbf9f9c359c4edb0349f9b)
 - Transport: BridgeTransport 2.0.33