trezor/trezor-firmware #3008: Text centering with TextLayout is not precise

TychoVrahe, 2023-05-10:
Some characters have more blank columns on one side then others, which can result in non-properly centered Label/Paragraph.

Centering of the TextLayout should take this into consideration.

Consider also how adding hyphen/arrow affect this, do we want centered text with hyphen or not?

(possibly, placing on start/end should also consider this).

Here is some idea how to approach this, but it requires more care: https://github.com/trezor/trezor-firmware/commit/ec9ad35e71e8f85a8a723bb890a5a9c8c014bc25

Hannsek, 2023-07-26:
@TychoVrahe is this related to T2B1 only?

TychoVrahe, 2023-07-26:
no its universal. but on T2B1 this is more visible due to low resolution

Hannsek, 2023-08-18:
E.g. double pixel space around letter I
[image]