trezor/trezor-firmware #3015: T2B1 - internal error screen

Hannsek, 2023-05-15:
[image]

Hannsek, 2023-07-26:
@matejcik Can you tell what actions can cause internal errors?

matejcik, 2023-07-26:
none, hopefully :woman_shrugging: 

you'll only see internal errors in case of (a) hardware failures or (b) some improperly handled bug or something

Hannsek, 2023-07-26:
Can we just change the design of error screen even if we don't know how to reproduce it?

matejcik, 2023-07-26:
sure.

Hannsek, 2023-08-17:
@TychoVrahe please provide us with current screenshot

TychoVrahe, 2023-08-17:
[image]

Hannsek, 2023-08-17:
Aha, I saw some different design. So if we have the layout and it is low-hanging fruit, let's change it.