trezor/trezor-firmware #3016: T2B1: regenerate icons with odd height

grdddj, 2023-05-15:
In the past, there was a limitation that the icons need to have even height. Because of that, some icons have now an empty bottom row, to satisfy this limitation. We can regenerate those to make them odd-heighted

Hannsek, 2023-06-17:
Is this done also for bootloader?

grdddj, 2023-06-19:
> Is this done also for bootloader?

Bootloader is using the majority of icons from firmware, so it benefits from this as well.

There are some extra icons like these

[image]

that I cannot find as `png` icons (in `core/assetrs/model_r`). These work fine however, so no necessary action there

Hannsek, 2023-06-19:
Ok, eventually, we can put there pngs of these icons.

closed 2023-06-20 by grdddj