trezor/trezor-firmware #3018: Improve plausible deniability when using SD card encryption

dnaleor, 2023-05-16:
Is your feature request related to a problem? Please describe.

When using SD card encryption, anyone that get hold of your device can know this. If you don't insert the card and you boot up the device, an error message appears

Describe the solution you'd like

It would be useful if in this case, another wallet would be created, so you can use it as a decoy with a small amount. Just use the encrypted seed that's inside the trezor already as the master seed. The seed words for this wallet should be shown the first time you open this wallet. If people for some reason don't want this feature, you could use a flag of some sorts to keep the warning message instead.

Describe alternatives you've considered

I don't see any alternatives for getting better plausible deniability in this case

wpalczynski, 2023-05-23:
This is quite a dangerous oversight given some countries can compel you to provide your encryption private keys under pain of jail.  Seems like quite an easy fix.  Any ETA on when this vulnerability will be adressed @Trezor?

Hannsek, 2023-05-23:
No ETA on this. However, you can still use standard wallet as a decoy and have majority of your funds in hidden wallet (passphrase protected).

prusnak, 2023-05-23:
SD card encryption has one goal: improve security against physical attacks.

Plausible deniability is not goal of this feature. If the feature had plausible deniability, this would introduce really bad user experience, since users would be super confused what's wrong with their wallet.

If you want plausible deniability, there is already a tool for that: the passphrase.

closed 2023-05-23 by prusnak

dnaleor, 2023-05-23:
@prusnak when traveling and you get searched, or when a burglar finds your trezor, it would be better to have this plausible deniability. And the confusion can easily be fixed with some kind of flag for users who want to opt out. Also, you need the command line tools anyway to get SD card encryption, so at the moment, only "power users" are using SD card encryption anyways...

wpalczynski, 2023-05-23:
Good.

prusnak, 2023-05-23:
> @prusnak when traveling and you get searched, or when a burglar finds your trezor, it would be better to have this plausible deniability

I repeat: You have plausible deniability using passphrase.