trezor/trezor-firmware #3019: T2B1 homescreen upload

Hannsek, 2023-05-17:
I tried to upload 
[image]
onto the device.

I got this: 
[image]

However: this [image] went well. (Picture from https://trezor.github.io/homescreen-editor/)

grdddj, 2023-05-29:
The issue is in the uploaded image, it does not have `black(0,0,0)`, but `black(25,25,25)`, so it makes the "dotted" background. 

With a completely black/white image I had no issues (and I guess the hedgehog is also purely b/w)

Try this screenshot:

[image]

Hannsek, 2023-05-29:
Shouldn't  trezorlib convert it to black and white automatically?

matejcik, 2023-05-29:
how would trezorlib know which color to assign to black?

grdddj, 2023-05-29:
> Ok, how it is done [here](https://trezor.github.io/homescreen-editor/)?

Looks like the `brightness` option is controlling the threshold for black/white.

Something similar _could_ be easily done by `trezorlib`, but I do not think we want to do these things:

```python
def black_or_white(image_path, threshold=128) -> Image:
    # Open the image file
    image = Image.open(image_path)
    # Convert image to grayscale
    image = image.convert('L')
    # Apply the threshold
    return mage.point(lambda p: p > threshold and 255)
```

matejcik, 2023-05-29:
yeah, trezorctl is not an image editor.
why is it a problem to supply an image with the right colors?

closed 2023-05-31 by grdddj