trezor/trezor-firmware #3021: Add option to Lock the current firmware version (the device stops accepting further firmware updates)

prusnak, 2023-05-19:
User story:
- user does not want to update firmware on their device never again
- they select the "lock firmware version"
- they confirm PIN
- device sets the flag
- from now on any attempt to update firmware on the device fails

Hannsek, 2023-05-19:
What is the motivation behind not updating the fw ever?

prusnak, 2023-05-19:
> What is the motivation behind not updating the fw ever?

After the current Ledger fiasco some people expressed the desire to have such feature.

mcudev, 2023-05-20:
Seems like theatre since the boardloader will allow overwriting the bootloader. Granted the bootloader would need to be SL production signed. Still reduces down to trusting SL protecting the signing keys and never being evil.

prusnak, 2023-05-20:
> Seems like theatre since the boardloader will allow overwriting the bootloader.

There is a bootloader downgrade protection.

TychoVrahe, 2023-05-20:
The issue is with bootloader upgrade (not downgrade) via SD card, where Satoshi Labs would issue newer signed bootloader, which would again ignore firmware lock version. 

For new trezor generations i think we should consider option (e.g. via OTP flag) to disable overwriting bootloader from SD card, which would allow effective implementation of this freeze feature, as well as pin protected firmware updates, which is something thats probably useful for more people than complete freeze.

mcudev, 2023-05-20:
Boardloader doesn't care: https://github.com/trezor/trezor-firmware/blob/master/core/embed/boardloader/main.c#L117

uj, 2024-09-25:
YubiKey does this, so this is not an unrealistic request.  **Actually, YubiKey NEVER allows firmware updating, if you want new firmware, you have to purchase a new device.  YubiKey is extremely popular, and this feature is one of their selling points.