trezor/trezor-firmware #3034: Device name does not support diacritics

iBobik, 2023-05-24:
**Describe the bug**
After update to 2.6.0 my name on the Trezor was crippled.

**Firmware version and revision**
2.6.0

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite
 - OS: macOS
 - Version 23.4.2

**To Reproduce**
Steps to reproduce the behavior:
1. Name your device to e.g. "T žščřěýáíéúů 😀"
2. See inverse `?` characters on the Trezor screen

**Expected behavior**
All UTF characters will be rendered properly

**Screenshots**

[image]

[image]

gat68, 2023-05-24:
> i don't think this had *ever* worked, so not sure why you think the
> update changed anything?
>
> I am not sure. It is possible it never worked.

closed 2023-05-24 by Hannsek

Hannsek, 2023-05-24:
Yeah, I have this in my mind for several month. We would eventually solve it in suite, where we wouldn’t allow diacritics and special characters.
I closed this issue as there is no way in near future we would implement this.

Hannsek, 2023-05-31:
Added the [issue](https://github.com/trezor/trezor-suite/issues/8604) to Trezor Suite.