trezor/trezor-firmware #3040: Deterministic build of `legacy/bl1.11.0` fails.

brianddk, 2023-05-27:
**Describe the bug**
Deterministic build of `legacy/bl1.11.0` fails.  When performing a `build-docker.sh` , the build of the previous (not current) bootloader fails to match checksums.

**Firmware version and revision**
Builds were run from Docker on Windows under WSL.  Since the build of `legacy/bl1.12.1` produced matching checksums, my confidence in my build environment is high.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone trezor-firmware && cd trezor-firmware`
2. `git checkout legacy/bl1.11.0`
3. `build-docker.sh --skip-core --skip-bitcoinonly legacy/bl1.11.0`
4. `sha256sum build/legacy/bootloader/bootloader.bin legacy/firmware/bootloader.dat`

See: https://github.com/brianddk/attestation for more detail

**Expected behavior**
When replacing `legacy/bl1.11.0` with `legacy/bl1.12.1`, the above four steps produce expected results.

**Screenshots**

The checksum for `legacy/firmware/bootloader.dat` is `0283ec8ca08064e6b4a4dadac01b2777c9bc08334a5e0bf39a6f9e6ae08ef788` which is what I'm trying to reproduce.  For completeness, I ran `docker-build.sh` with a number of commit tags near `legacy/bl1.11.0` just in case.  For example:

```
build-docker.sh --skip-core --skip-bitcoinonly 78feb5fe0
```

Here are the checksums I got, none of which are what I was trying to achieve.

|**Git TAG**| **sha256sum of build/legacy/bootloader/bootloader.bin**        |
|:---------|:----------------------------------------------------------------|
|85a26d2c9 | 3f5c6f172c9553dcea34bc61c08172ceaa4e1767d700e14b5f1479486731def1|
|cc4ca7a58 | 3f5c6f172c9553dcea34bc61c08172ceaa4e1767d700e14b5f1479486731def1|
|a1984f327 | 3f5c6f172c9553dcea34bc61c08172ceaa4e1767d700e14b5f1479486731def1|
|768d577d7 | 3f5c6f172c9553dcea34bc61c08172ceaa4e1767d700e14b5f1479486731def1|
|bc1ebdccd | b3f5397d80bb9b7e4233d4c4dbb442b109ebedab02790d4cf16b133725b4e1e1|
|ed8cdef33 | b3f5397d80bb9b7e4233d4c4dbb442b109ebedab02790d4cf16b133725b4e1e1|
|7d93bd317 | 8a5b5691127bb6a7201dfdd066f152e11cce51037034e4ff0afb762430df367a|
|29bab9e8a | 37fc2e91a5a8c514f4eb3e26eb615b439b51c968bbc5d87e26d1b8318881fc7f|
|78feb5fe0 | 37fc2e91a5a8c514f4eb3e26eb615b439b51c968bbc5d87e26d1b8318881fc7f|

**Additional context**

Much of this was also discussed in https://github.com/trezor/trezor-firmware/issues/2189

brianddk, 2023-05-27:
Disregard last.

The problem was indeed in my build system, much to my chagrin.  To resolve my environmental issues I did the following:

1. Purge ALL docker containers, images, and settings
2. Remove the `trezor-firmware` repo
3. Checkout DIRECTLY into proper tag: `git clone -b legacy/bl1.11.0 https://github.com/trezor/trezor-firmware`
4. Run build: `cd trezor-firmware && ./build-docker.sh --skip-core --skip-bitcoinonly legacy/bl1.11.0`

After this, checksums match.

closed 2023-05-27 by brianddk

matejcik, 2023-05-29:
yeah, one problem of the current "reproducible" process is that you need your "outer" repo to be on the right commit, and also pass that same tag to the "inner" docker environment. changes in the outer script can affect results of the inner one.

perhaps the inner script should just verify this?

brianddk, 2023-05-29:
Well my current audit is pretty favorable.  I've been able to build everything for the last two years.  What's more, each build does indeed have a magic commit that works both outside and inside docker environment.   I'll push my findings in the next couple of days and update.  Worst case I could imagine is to add some tag like `core/bl2.1.0.build` or the like to notate that there is one or two "special" cases.