trezor/trezor-firmware #3047: Behaviour of the xpub displaying should be aligned with address displaying

Hannsek, 2023-05-31:
https://github.com/trezor/trezor-suite/issues/3454#issuecomment-1569917006

Hannsek, 2023-06-01:
Probably related: https://github.com/trezor/trezor-firmware/issues/3043

bosomt, 2023-06-07:
Checked it with latest version 23.6.1 that will be soon released and it does not show complete xpub address when address is too long.
problematic coins are:
Cardano Ada and Bitcoin taproot address.

**Model One** via confirm button you are navigated to rest of XPUB characters
**Model T** shows first part of xpub and rest is ignored so if you only check last charasters of XPUB it wont match when compared with Trezor Suite.

[image]

mmilata, 2023-07-24:
I can confirm the Cardano issue but not the taproot one. Can somebody please provide steps to make trezor display truncated taproot XPUB?

The old UI did not show QR code for XPUBs, I'll see if it can be added such that it's scanable.

bosomt, 2023-07-26:
Checked it one more time /multiple accounts + testnet/ and Bitcoin taproot XPUB is displayed **correctly** /except prefix missing / sorry for confusion. 

```
tr([2c27fa4b/86'/0'/0']xpub6BvdKTJruDymKUqK7wm9q3NsieBnNzjeYEC6ik25wPLY5aR3A34TQhJtfnegRTZmWPbcaBbB1NphhnfiSz3EXmU4uKqkx9BaiCr9d8q5DSS/<0;1>/*)

vs

xpub6BvdKTJruDymKUqK7wm9q3NsieBnNzjeYEC6ik25wPLY5aR3A34TQhJtfnegRTZmWPbcaBbB1NphhnfiSz3EXmU4uKqkx9BaiCr9d8q5DSS

```

PR #3177 (2023-09-11): Display QR codes for XPUBs
Fixes #3047.

Also includes fix for Cardano to display the same hex data (including the chain code) that it sends over protobuf.

[master diff](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4746721205/artifacts/master_diff/master_diff.html)

[image] [image] [image] [image] [image] [image]
closed 2023-09-11 by mmilata