trezor/trezor-firmware #3057: Older 2021 Trezor deterministic builds are now failing

brianddk, 2023-06-03:
**Describe the bug**
Current (2022-2023) deterministic builds of Trezor firmware and bootloader are working as expected, but the core builds from 2021 seem to be failing.  Legacy builds from this period still seem to be deterministic.  Of course much older builds from 2020 can no longer create the build environment to even attempt the build, so are not even considered for review.

**Firmware version and revision**
Varies

**Build Environment**
 -  Windows 10: 22H2 v10.0.19045 Build 19045
 - WSL v2.0 serving Ubuntu 22.04
 - Compiled Kernel v5.15.90.1-microsoft-standard-WSL2+ with HIDRAW enabled
 - Docker Desktop for Windows (and WSL) v4.19.0

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the procedure in [reproducible-build.md](https://github.com/trezor/trezor-firmware/blob/core/v2.6.0/docs/common/reproducible-build.md) (`build-docker.sh core/v2.3.6`)
2. Use the commit tag `core/v2.3.6` to observe a clean build
3. Follow the procedure in [reproducible-build.md](https://github.com/trezor/trezor-firmware/blob/core/v2.6.0/docs/common/reproducible-build.md) to compare binaries
4. Notice the binaries fail the compares after following the procedure outlined.

**Expected behavior**
The [reproducible-build.md](https://github.com/trezor/trezor-firmware/blob/core/v2.6.0/docs/common/reproducible-build.md) should work 2.3.6 and 2.3.5 just as it does on 2.6.0 and 2.5.3.

**Sightings**
- [Github : brianddk : attestation : failed : core : v2.3.6](https://github.com/brianddk/attestation/blob/73aaf53/trezor/failed/core/v2.3.6/attest.brianddk)
- [Github : brianddk : attestation : failed : core : v2.3.5](https://github.com/brianddk/attestation/blob/73aaf53/trezor/failed/core/v2.3.5/attest.brianddk)

**References**
- [Github : brianddk : attestation : failed : core : v2.3.6 : log](https://github.com/brianddk/attestation/blob/73aaf53/trezor/failed/core/v2.3.6/build-log.brianddk) (downloadable)
- https://github.com/brianddk/attestation/issues/8
- https://github.com/brianddk/attestation/issues/9

prusnak, 2023-06-03:
I don't think we can't do anything there, because Debian Docker builds are not guaranteed to be stable.

That's why we migrated to NixOS which is guaranteed to be reproducible (forever).

brianddk, 2023-06-05:
> I don't think we can't do anything there, because Debian Docker builds are not guaranteed to be stable.

But 2.3.6 was already using NIX, if I'm reading the Dockerfile correctly:

https://github.com/trezor/trezor-firmware/blob/b19cbf67c6c7c38513947b703df6d4409c59bc98/ci/Dockerfile#L3-L15

b19cbf6 is what I see tagged for 2.3.6