trezor/trezor-firmware #3058: Can't upgrade from <1.12.0 to 1.12.1, Trezor 1

andrewbaxter, 2023-06-04:
**Describe the bug**
I'm trying to update my Trezor after not using it for a while.

```
$ trezorctl firmware update
Best available version: 1.12.1
Version 1.12.0 is required before upgrading to 1.12.1.
Version 1.12.0 not found for your device.
Closest available version: 1.12.1
$ trezorctl firmware update -v 1.12.0
Version 1.12.0 not found for your device.
Closest available version: 1.12.1
```
```
$ trezorctl version
0.13.7
```
Python 3.11

This seems like a release availability issue so I think the above details are what's required.  I searched a bit and couldn't find the error message in bug reports or google (generally).

Hannsek, 2023-06-04:
As the version is <`1.12.1` you have to install intermediate firmware first: https://data.trezor.io/firmware/1/trezor-inter-v2.bin

andrewbaxter, 2023-06-04:
Is that documented anywhere? And shouldn't trezorctl identify and install that?

andrewbaxter, 2023-06-04:
That didn't go well - I can't read my screen well due to lcd burnout from the pre-screen-blank patch a long time ago so bear with me but:
1. I do the update
2. It asks me to confirm the hash (where can I go to confirm the hash?)
3. It says install successful and reboots
4. It says "unofficial firmware detected"
5. It asks me to confirm the hash again
6. It says "Cannot update" "Unpriviledged mode" "Unsigned firmware"

andrewbaxter, 2023-06-04:
I'm having a hard time believing this is a question and not a bug... the official upgrade process shouldn't go like this.

Hannsek, 2023-06-04:
It indeed should not go like this. Please contact Trezor support where they will help you with the update.

andrewbaxter, 2023-06-04:
Ah okay, so to be clear you think there's something unusual about my setup causing issues and this isn't a typical result for T1 upgrading to 1.12.1.

Just so I have the right context, normally would `trezorctl` upgrade a T1 from 1.11.1 to 1.12.1 without issues or needing the `trezor-inter-v2.bin` file?

Hannsek, 2023-06-04:
You can use Suite for guiding you through the update process.

> Just so I have the right context, normally would trezorctl upgrade a T1 from 1.11.1 to 1.12.1 without issues or needing the trezor-inter-v2.bin file?

If by _normally_ you mean standart fw update, than yes, you won't need any intermediate fw.

andrewbaxter, 2023-06-04:
By normally I mean for people with properly functioning hardware, assuming this is some sort of hardware failure issue... what do you mean standard fw update?  `trezorctl firmware update` is the standard firmware update, right?

I just tried this with a different T1 though and got the same
```
$ trezorctl firmware update
Best available version: 1.12.1
Version 1.12.0 is required before upgrading to 1.12.1.
Version 1.12.0 not found for your device.
Closest available version: 1.12.1
```
issue.

andrewbaxter, 2023-06-04:
Okay, 1st I had to wipe but then updated directly to 1.12.1 okay. I couldn't recover because the screen was too burnt in to read the prompts reliably.  2nd T1 took the inter- firmware, but got stuck in bootloader mode.  Ended up having to wipe and recover that one too.

The inter- firmware worked for neither and `trezorctl` automatically updated to 1.12.1 on neither; this looks, smells, and sounds like a bug but maybe it isn't a bug.  Either way, I've done my job reporting it here.

matejcik, 2023-06-12:
there's two issues at play here:

(a) trezorctl doesn't auto-support intermediate fw. i thought there was an issue about it but perhaps not?
(b) the glitch with 1.12.1 -> 1.12.0 -> 1.12.1

we definitely want to solve (a), i thought there's even an issue for that, but can't find it right now

as for (b), that is a weird corner case that shouldn't even occur when we do solve (a), so while that's like technically a bug, it's probably not worth the effort of fixing