trezor/trezor-firmware #3076: TOS link not visible in Shamir backup wallet creation

grdddj, 2023-06-14:
For `TT`, when choosing (Super)Shamir backup, the wallet creation screen does not show `trezor.io/tos` link as it cannot fit the screen because of two-line title

Mentioned at https://github.com/trezor/trezor-firmware/pull/3074#issue-1756683383

[image]

Hannsek, 2023-06-18:
With new wording, `SHAMIR` should fit on one line. 
[image]

However, `SUPER SHAMIR` wouldn't fit on one line. Is it possible to have different vertical space between the "paragraphs"? 
[image]

grdddj, 2023-06-19:
> With new wording, `SHAMIR` should fit on one line. [image]

Right, we are currently hardcoding the line-break, we can just replace it by normal space

 
> However, `SUPER SHAMIR` wouldn't fit on one line. Is it possible to have different vertical space between the "paragraphs"? [image]

I guess it will be possible to get rid of the vertical space when the label has two lines - we are currently putting `\r` at the end of the first paragraph to create the space

prusnak, 2023-06-19:
I think we should drop "SUPER SHAMIR" and just use "SHAMIR" for this case as well. We don't use this wording anywhere in the product.

grdddj, 2023-06-19:
> I think we should drop the word "SUPER| from "SUPER SHAMIR" and just use "SHAMIR" for this case as well. We don't use this wording anywhere else in the product.

That would make things easier, I am for it :)

Hannsek, 2023-06-19:
Fair enough.

closed 2023-07-18 by grdddj