trezor/trezor-firmware #3087: Vertically center character input carousel - pin

Hannsek, 2023-06-17:
https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/4488853768/artifacts/test_ui_report/passed/TR-click_tests-test_pin.py%3A%3Atest_pin_change.html

[image]
[image]

Should be centered between the upper character line 
numbers/text) and the bottom button.

Hannsek, 2023-06-17:
We can do either this or do it the same as passphrase input. i.e. header would be still present and actual pin entries would be below it.
[image]

Hannsek, 2023-06-17:
The second option is probably better. What do you think @grdddj?

grdddj, 2023-06-19:
> The second option is probably better. What do you think @grdddj?

I like this, so it will show `ENTER PIN/WIPECODE` all the time. The second line will be then a subprompt (`You have 15 tries left`) when PIN is empty, or asterisks when user already inputted something.

grdddj, 2023-06-19:
> `You have 15 tries left` is visible only after the first unsuccessful try, no?
> 
> [image]

Right, not on the first try. We currently do not show the underscore when the PIN is empty, but will do it

Hannsek, 2023-06-19:
I'd say to that one below, as the user wants to see the words. 
[image]

grdddj, 2023-07-17:
Merged in https://github.com/trezor/trezor-firmware/pull/3135

closed 2023-07-17 by grdddj