trezor/trezor-firmware #3089: Wallet creation, seed backup

Hannsek, 2023-06-18:
https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?type=design&node-id=8825-71197&t=4gMBHfDOelVRGYl9-4

Hannsek, 2023-06-19:
Thanks for that. 

As pagination on the `STANDARD BACKUP` page would "ruin" the header, let's not put it there.
[image]

> Also, with the left-arrow, just want to confirm (as I see it in other designs as well) that we have changed its appearance - it is always a "slim" arrow instead of the "fat" arrow as before?

Yes, with the one-fat-arrow design, the user is more guided in which direction he/she should continue.

grdddj, 2023-06-19:
> But user will be able to go up to the `STANDARD BACKUP` page by the up arrow. Is it ok from the code pov? @grdddj

I think it is fine implementation-wise.

To which point would the user be redirecting when selecting `CHECK AGAIN` after an unsuccessful "quiz" attempt?

Hannsek, 2023-06-19:
I'd say to this one below as the user wants to see the words. 
[image]

grdddj, 2023-06-26:
> I'd say to this one below as the user wants to see the words. [image]

For simplicity purposes, currently, it moves to the previous page (`Write down all XX words`) - it is an edge case (caused by user error) and it would unnecessarily complicate the code to move in the middle of one flow

Hannsek, 2023-07-04:
Wallet creation - Shamir backup design: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Firmware?type=design&node-id=8825-71198&mode=design&t=S14phM8hFc2KN54k-4

grdddj, 2023-07-17:
Merged in https://github.com/trezor/trezor-firmware/pull/3135

closed 2023-07-17 by grdddj