trezor/trezor-firmware #3090: PIN

Hannsek, 2023-06-18:
https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?type=design&node-id=8825-71199&t=4gMBHfDOelVRGYl9-4

grdddj, 2023-06-19:
[image]

Here, we would currently show the last inputted digit - `**3` - is it OK, or the inputted `3` should only be visible right after choosing it, and every subsequent interaction should hide it?

Hannsek, 2023-06-19:
I think it is better tho hide it with every subsequent interaction. What do you think?

grdddj, 2023-06-19:
> I think it is better tho hide it with every subsequent interaction. What do you think?

Agreed, it is safer and I hope will be enough confirmation for the user that the inputted digit is there. Implementation-wise it looks doable

grdddj, 2023-06-20:
I am missing updated screens/copy for PIN disabling and PIN change.

Also, wipe code specification is not there.

grdddj, 2023-06-20:
> Can you please provide me with desired screens that should be changed?

Sure, these are current texts and buttons for PIN and wipe-code that do not have Figma designs:

```
description="Do you want to disable PIN protection?",
verb="DISABLE",

description="Do you want to change your PIN?",
verb="CHANGE",

description="Do you want to disable wipe code protection?",
verb="DISABLE",

description="Do you want to change the wipe code?",
verb="CHANGE",

description="Do you want to enable wipe code?",
verb="ENABLE",
```

grdddj, 2023-07-17:
Merged in https://github.com/trezor/trezor-firmware/pull/3135

closed 2023-07-17 by grdddj